hamishcunningham/pi-tronics #27: Any new versions in the works?

gregcope, 2015-01-03:
Is there any chance of newer versions?

For example being able to charge one of the inputs from the other (think solar on no1 charging a battery on no2)?

External wake on signal?

hamishcunningham, 2015-01-04:
hi Greg,

There is a new board in design; charging was a priority, but I don't think we discussed wake on signal -- we're concerned to minimise the current draw when the Pi is not running.

Not sure when the new version will surface though -- we all got busy after the summer :-(

Closest thing I can think of is circuits like this: https://www.circuitlab.com/circuit/shevrt/in-car-mopi/
which uses external switching to track a separate 12V line... Any use?

Best

h

gregcope, 2015-01-04:
Minimising current draw is higher priority than wake on signal for me too.

I am just about to buy a few more, so sounds like, in this instance, I will go with the present version and see what surfaces later.

Thanks.

hamishcunningham, 2015-01-04:
cool, have a good one
h

closed 2015-01-04 by hamishcunningham