hamishcunningham/pi-tronics #28: MoPi with Solar and Batteries for floating PoolPi PC

wayner9, 2015-01-03:
I am planning to make a Pi based PC that will float in my swimming pool and give me readings of water temperature, etc.  I plan to use solar panels and rechargeable batteries to power a Model A+ with wifi and I may add more peripherals (like a camera) depending on power consumption.  The solar panels should also be able to charge the batteries during the day when the sun is shining.

Does anyone have any advice on using Solar Panels in conjunction with rechargeable batteries and a MoPi?  There is discussion of solar panels on the MoPi page but the main setup seems to be using mains+solar rather than batteries+solar.  Should I have similar voltage of the bank of batteries and the solar panel?  For example, if I have a 12V solar panel should I put 8 AA rechargeable batteries in series?  These should give me voltage of about 10-11V since it seems that rechargeable batteries don't give you the "full" 1.5V.  And could I then connect this up to the inputs of the MoPi?

Would I need a voltage or current regulator at all between the solar cells and the batteries?   Or is this not needed if they have similar voltage - that is kind of implied by this page http://www.reuk.co.uk/Solar-Battery-Charger-With-LM317T.htm

How about a diode arrangement of some sort to ensure that current only flows from solar cell to batteries and not the other way around?

Any idea how long I would get from 8 fully charged Duracell AA NiMH rechargeable batteries?

Any reason not to use a MoPi with solar panels and rechargeable batteries in this application?

hamishcunningham, 2015-01-04:
> I am planning to make a Pi based PC that will float in my
> swimming pool and give me readings of water temperature, etc.
> I plan to use solar panels and rechargeable batteries to power
> a Model A+ with [...]

Sounds like a good project!

> Does anyone have any advice on using Solar Panels in
> conjunction with rechargeable batteries and a MoPi? There is
> discussion of solar panels on the MoPi page but the main setup
> seems to be using [...]

If I understand correctly you want to power the Pi from the
panels when they are getting enough light... in which case you
want to arrange the battery voltage to be lower than the panels
at that point.

> Would I need a voltage or current regulator at all between the
> solar cells and the batteries? Or is this not needed if they
> have similar voltage - that is kind of implied by this page
> http://www.reuk.co.uk/Solar-Battery-Charger-With-LM317T.htm

Yes, you need a separate charger for the batteries, and this is
not trivial to arrange because many batteries require charging
regulation for each cell. Probably best with e.g. a single lead
acid or LiPo setup. In any case this will also need external
switching to isolate it from the Pi when charging...

> How about a diode arrangement of some sort to ensure that
> current only flows from solar cell to batteries and not the
> other way around?

Yes, see example circuits here:
https://www.circuitlab.com/workbench/

> Any idea how long I would get from 8 fully charged Duracell AA
> NiMH rechargeable batteries?

There's loads of data at https://pi.gate.ac.uk/pages/mopi.html#detailed-testing
(I've not seen Duracell NiMH though -- the key thing is how many
mAH they supply...)

> Any reason not to use a MoPi with solar panels and
> rechargeable batteries in this application?

Nope! Good luck, H

jeromeavot, 2015-01-05:
Hello,

Nice project indeed ! You could also consider using a dedicated pool sensor
such as the Oregon THWR800 which can then be easily read at distance using
a Raspberry Pi (using Heyu for instance) and a RFXCOM Module.

I'm using the same kind of system for monitoring my house temperature (it
is not THWR800 but THGR810 but the idea is the same.

Just let me know if you need any additional information.

Jérôme.

hamishcunningham, 2015-01-05:
good idea jerome, tnx!
h

wayner9, 2015-01-05:
Thanks for the responses folks!

Hamish - the common Duracell rechargeables sold here in NA are NiMH - these are mainly green with a "copper-top".  I think the ones sold in the UK are as well.  

What do you mean by "external switching to separate it from the Pi while charging"  Does that mean that I could not connect the solar panel in parallel with a lead acid battery and connect this to the power leads of a MoPi?  Can you point me to an example of this type of switching?

Doing more reading it seems like sealed lead acid may be the way to go as, from what I have read so far, they are less finicky to charge.  My security system also uses these types of batteries and I have a spare one on hand.  And these batteries are used in UPS units so they are readily available, inexpensive and come in a variety of sizes.  They are generally 12V - but there voltage goes a bit higher when fully charged.  Does that mean that I will need solar panels that are at least 15V?  Or would 12V solar panels suffice?  It looks like 12V is a common size for both so it would be nice if I could get away with 12V solar panels but that might just leave the battery half full.

@Jerome - That product seems to be discontinued.  But even if it was available I still want the challenge of making my own devices, especially since I would like to add functionality.  In an ideal world it would measure temp, free chline, pH, Total Alkalinity and Calcium Harness.  But most of these can't be measured electronically, at least not cost effectively for an application like this.

wayner9, 2015-02-02:
Here is a $15 solar panel controller that I found on the net. http://www.instapark.com/solar-power-charge-controllers/ss1203-3amp-solar-power-charge-controller-45-watt-at-12v.html  This might be the easiest way to integrate solar and batteries although I would imagine that only the battery would be hooked up to the solar panels.

If you are thinking about new versions of the MoPi - how about one that combines a solar charge controller with the current MoPi?  One of the inputs could be for solar panels and the other for battery. It would be great if you could read both voltages and keep an eye on the data coming from the unit and check the status of charging.

hamishcunningham, 2015-02-04:
good ideas -- the next version has a connector for add-on charging modules,
to cater for various inputs (inc. solar) and outputs (lipo, lead/acid, ...)
great minds think alike :-)
(but also: idiots seldom differ! the future will tell us!)
cheers
h

wayner9, 2015-02-04:
Any idea when the next version will be available for purchase?

hamishcunningham, 2015-02-17:
Next version: we've done a lot of the design work, and Lubo has put together a first cut prototype board layout... But then we got busy :-(

So I'd say a few months before it is ready for prime-time.

closed 2015-02-17 by hamishcunningham