hamishcunningham/pi-tronics #33: Any way to power on Pi with MoPi when Vin>6.2V?

wayner9, 2015-02-02:
I see that there is a timer included with MoPi that will wake up your Pi at a specific time.  Is there any way to do this by Input Voltage rather than time?  It would be nice to have a solar powered unit that can shut down when the light is insufficient but that can then come back up when there is suffficient light to generate > 6.2V.

wayner9, 2015-02-15:
Bump - is this possible?  It would be great functionality to have, especially when one has a project running from solar or solar charged batteries.

hamishcunningham, 2015-02-17:
A circuit like this one should do the trick: https://www.circuitlab.com/circuit/shevrt/in-car-mopi/

I blogged a bit about this here: https://pi.gate.ac.uk/posts/2015/02/11/in-car-mopi/

HTH, best

Hamish

closed 2015-02-17 by hamishcunningham