hamishcunningham/pi-tronics #35: Voltage measurement starts at 5.07V - why is this?

wayner9, 2015-02-03:
I have placed an A+ with a MoPI, a battery on one MoPi input and 2x3V solar cells (connected in series) on the second input.  This Pi is in my greenhouse where the ambient temperature is very low right now - around -12C plus any heat generated by the sun.  It looks like I get a reading of 0 from the input voltage of the solar cells until there is enough light to get the solar cells to 5.07V.  Then I get valid readings until later in the day when the sun starts to set.

Is there a technical reason why voltage measurement starts at this level?  Or was this a design choice?  It might be nice to also read lower voltage levels when using solar cells.

hamishcunningham, 2015-02-04:
the specified lowest operational voltage of mopi is 6.2V -- results that
you might get below there are unpredictable!
best
h

wayner9, 2015-02-04:
Ok, thanks Hamish.  My 15V solar cells arrived from China last night.  I will be writing some code to log the voltage readings to a database and report through a web UI. Do you think this is something that other folks will find useful?  If so where should I post it?  Here at Github? Or is there a more centralized place where people can find MoPi related project?

hamishcunningham, 2015-02-04:
yes, sounds great; if you want to do a draft in .mkd format I can post it
on pi.gate.ac.uk ?
best
h

wayner9, 2015-02-04:
I am not familiar with mkd - I was going to have python code and a bit of textual documentation.  

My code will be based off of these two web pages but will focus on logging voltage rather than temperature:
http://raspberrywebserver.com/cgiscripting/rpi-temperature-logger/building-an-sqlite-temperature-logger.html
http://raspberrywebserver.com/cgiscripting/rpi-temperature-logger/building-a-web-user-interface-for-the-temperature-monitor.html

hamishcunningham, 2015-02-04:
cool -- mkd is markdown, which is used by github and others

but any format is ok really :-)

just let me know what you;d like to do!
best
h

closed 2015-02-17 by hamishcunningham