hamishcunningham/pi-tronics #36: device tree breaks I2c

achrn, 2015-02-07:
Device tree in new firmware means that I2C in the existing software for Mopi does not work.  Furthermore, the error at install time that would tell you this gets overlooked (by me at least) because the web page says "you will get a warning message; just ignore it at this point!"  You do get a warning message, and it's very similar but not quite the same as the warning message you should ignore.

Anyhow, I don't know if this is easily fixed in the software to configure the I2C automagically, but it's easy to fix by manual intervention when you know how:

sudo raspi-config

select 'advanced options'

select 'I2c' and say you do want to enable the I2C interface, and you do want to load the module

Then everything works.  Adding this sequence to the troubleshooting section on the web site would be good.  Putting something against the 'you'll get a warning message just ignore it' section would help too.

hamishcunningham, 2015-02-09:
hi,

Thanks for this!

Are you running a B+ or...? I haven't seen this error as yet -- though RPI.GPIO seems to be broken on the Model 2 which scuppers things :-(

(As Ben said, why didn't they give developers some warning?!)

cheers

h

achrn, 2015-02-09:
Yes, 'just' a B+, though one with a bleeding-edge 'sudo raspi-config', so I don't think it's an issue yet in the less exciting apt-get update / apt-get upgrade cycle.

It's nothing to do with the Pi2-breaks-gpio issues.

hamishcunningham, 2015-02-09:
ok, tnx, will try and duplicate

jdotbdot, 2015-02-15:
I get the same issue with models B and B+.

MoPi works on a freshly installed system using 2014-12-24-wheezy-raspbian.img
It stops working after apt-get update and apt-get upgrade

It is fixed by re-enabling I2C in raspi-config.

hamishcunningham, 2015-02-17:
I have fixed this and made a new release (4.1) to the PPA -- if you get a chance to try it out that would be great :-)

Now updating the docs...

Thanks for the heads-up, sorry it took me a while (I haven't rebuilt the package since last June, and it is a bit of a complex beast...).

closed 2015-02-17 by hamishcunningham

jdotbdot, 2015-02-18:
Thanks for the quick response Hamish.

Unfortunately I can't test it because I have fried my MoPi by means of a
loose wire and short circuit.

I guess it's all part of the learning process!

Kind regards,
jb

hamishcunningham, 2015-02-18:
:-(

worth checking if it is just the fuse? (which should autoreset after a min
or two if not totally fritzed)

h

achrn, 2015-02-19:
Hi Hamish.  Thanks for the update.  Slight delay because my 'spare' pi was running a 3-day timelapse gathering process.  I obviously need a spare spare.

Anyhow, update is now in wherever it needs to be because I wrote a new flash card from the image that previously failed and after a new apt-get upgrade 30 minutes ago it all works first time with no problems.

hamishcunningham, 2015-02-19:
grand, thanks!
best
h