hamishcunningham/pi-tronics #37: What do the status bits mean?

wayner9, 2015-02-15:
From the CLI you can get a status reading.  But I didn't see any mention of how to decipher the status readings in the documentation.  

From some experimentation it looks like bit 0 = 1 if V2 > V1.

I see a lot of status readings of 4161(100001000000) and 4162 (1000001000001) and recently I have seen 20545 (101000001000001) - these are when everything is working well and I am running on mains power with a battery on one mopi input 1 and a solar cell on another mopi input 2.

I have also seen readings of 64 (just before my system shut down, likely due to cold temperature), 65, 832, 65407, 16192 5031, and 5070.

stoduk, 2015-02-15:
"mopicli -sv" should give you the decoded status - or look in the file
below to see the meaning of each bit.

HTH,
Anthony

https://github.com/hamishcunningham/pi-tronics/blob/master/simbamon/simbamond.default

wayner9, 2015-02-15:
Thanks Anthony - the file had all of the info that I was looking for.  Is there a way to tell if the Pi is running from mopi or mains?  

Bits 0 and 1 which tell you which power source are live seem to be active regardless of mains power being connected or not.

One other thing - somehow bit 14 appears to have been turned on.  How do I turn it off as it is unused?

stoduk, 2015-02-15:
IIRC bits 0 and 1 will just indicate if the supply is valid (ie. over
6.something volts that MoPi requires).

You mention mains power - but AFAIK all power should be coming via the MoPi
(connected to eg. a battery and a mains input).

Perhaps best if I leave this to Hamish and the team (I'm just a user) - but
if you clarify what your setup is then that might help them.

Cheers,
Anthony

wayner9, 2015-02-15:
Ok thanks Anthony.  I was under the impression that if you have mains connected (in other words powering via the Micro USB port on the Pi) that the mopi will not be providing power to the pi, but it is ready to take over if the mains power is lost.  This would be just like the battery on a laptop and would keep the pi up and running.

hamishcunningham, 2015-02-17:
Yes, that's true -- if the mains power is available it will overpower the supply from MoPi.

I'm closing this as I guess the Q was answered (thanks Anthony!); feel free to repoen if I've missed summut!

closed 2015-02-17 by hamishcunningham

wayner9, 2015-02-17:
Can I suggest that you add the status bits to the documentation page.

hamishcunningham, 2015-02-17:
Done: https://pi.gate.ac.uk/print/pages/mopi.html#faq-status-bits