hamishcunningham/pi-tronics #38: Wrong verbose status after reconnecting source

ISO-B, 2015-02-26:
I am running mopi with 12V DC adapter from wall and 10 x 1.2V 2200mAh rechargeable batteries. Sometimes I need to detach batteries and recharge them. After plugging them back verbose status for that source is: low/not present.

```
Firmware version: 3.10
Serial number: 387
Status word: 24646
Verbose status:
  Source #2 active
  Source full (blue led)
  Source #1 low/not present
  Source #2 good
  User configured
Current source voltage: 14235
Source #1 voltage: 13650
Source #2 voltage: 14235
```

stoduk, 2015-02-26:
[commenting as a user only]

Are these AA batteries in the usual cheap battery holder?

If you have a multimeter, what is the voltage across the batteries at the MoPi (on the screw terminals you have the battery attached to)?

Assuming these are AA batteries then you might be hitting the same headache I've had.  After some time  (it doesn't take long!) the springs just aren't up to the job of making a decent contact, so even though the batteries look like they are touching at the positive end it isn't a good enough contact to make a full circuit.  Sometimes everything is ok to start with (helped by pushing each battery in the direction the spring is trying), but the smallest knock displaces things and so kills power.

The right solution would be to find holders with better springs (or replace the springs in what you have).  The only fix I have so far is stretching the springs out - which I imagine I'll have to do periodically once they plastically deform again.

ISO-B, 2015-02-26:
I havent tested it with multimeter, but mopi tells voltage for that source. As you can see from my output mopi tells that there is voltage and verbose is wrong.

In my case Source 1 is for batteries and it has 13650 voltage. While batteries were detached mopi said voltage was something like 5XXX. After plugging batteries it jumped to 13650 before detaching batteries it was something like ~12500. So I dont think that springs are problem.

stoduk, 2015-02-26:
Ignore my previous comments about battery holders, this was a red herring.
Not enough coffee :)

[I've deleted the comments to avoid the distraction]

guruthree, 2015-02-26:
Is the amber LED for the power source flashing? If when the batteries were detached and the voltage reading didn't drop to 0, it's likely the status of the input wasn't reset. In this case then, it will report low as you are seeing. Try disconnecting the batteries and waiting longer for the voltage level reported to drop to 0. If you've already waited a while (e.g. the time for them to charge up) then a power cycle (disconnecting the wall source) should definetely sort it out.

ISO-B, 2015-02-26:
I will have to check that led when I can. Batteries where disconnected for 24h and it never dropped to 0.

What might prevent voltage reading going to 0 when source is detached?

Do you know does that have something else impact but that verbose is wrong? Example if I disconnect wall adabter will my pi shutdown or stay up until batteries run out.

guruthree, 2015-02-26:
With very high voltages (usually > 20V), there can be leakage across the schottky diode we use to combine the inputs. A quick fix _might_ be to disconnect the batteries and connect the input to ground. I'm not at home and can't test this myself, so this comes with a heavy warning that it may potentially damage your MoPi and/or RaspberryPi!

It's almost certainly the voltage not going to 0 causing the problems. If you run on batteries until the Pi shuts down, the voltage level is still there, so the microcontroller won't reset the status of the input. A full disconnect of all power sources will reset everything, of course.

hamishcunningham, 2015-02-28:
> Do you know does that have something else impact but that verbose is wrong? Example 
> if I disconnect wall adabter will my pi shutdown or stay up until batteries run out.

No, it won't create a problem with switching to another supply. There should be no break
in power to the Pi.

closed 2015-04-22 by hamishcunningham