hamishcunningham/pi-tronics #39: Mopi Rev 3 PCB "configuration"

Matioupi, 2015-03-25:
Hello, I can see a few hole and test points on the mopi (rev 3) PCB.
Is it possible to have a deeper description of them ?
Which resistors defines the voltage divider used to drive the TSP5430 to 5V ?
I would like to change this to 5.1/5.15V by modifying those, Is this possible by other means ?
Regards

Matioupi, 2015-03-25:
Ok, I got it :
R14 is a 10kOhm R15 is a 3090kOhm, so this gives à 1.221 (ref voltage of converter) * (13090/10000) = 5.17V minus the dropout of the SS14 schottky diode which gives slightly below 5V. When the pi2 is On, this give a 4.68 ~ 4.7V on the 5V rail at no load...

I've been soldering a 47kOhm in parallel of R15 and this way, I'm getting 4.95V on the 5V rail with pi2 connected and turned on.

On a next revision, it would be really nice to have a solder pad available / removable track / jumber to enable disable this 47k resistance in parallel to R15 and go for a "real" 5V supply when taking into account drop outs and everything...

I think I understand that I should not go with simultaneous micro USB powering in this configuration, but any other feedback from the conceptors on the modification I just did are very welcome.

LuboBonchev, 2015-04-22:
Hi Matioupi,

You did it great! Thank you. But please have in mind that the MoPi output voltage to Pi is depending on the particular power consumption. Due to the decoupling SS14 diode. That diode splits both 5V power supplies: MoPi and micro-USB. 
The next MoPi version contents so called "smart diode" which acts as a zero Ohm switch when is on. So, will be no drop on decoupling and the MoPi's voltage regulator will deliver 5V to the Pi (actually 4.9V - 4.95V again to give the ability to monitor the 5V at Pi site also). No additional tuning at the customer end will needed. 
Regards,
Lubo Bonchev
SelCom Ltd.

closed 2015-04-22 by hamishcunningham

Matioupi, 2015-04-27:
Hello,

do you have timeschedule for next gen MoPi ? and also preliminary
specifications ?

If you ask my guess for features I would love :

- 3 inputs instead of 2
- ~4A power capability (2 for pi + 2 for external devices e.g. screen)
- pcb holes for remote installation of power status LED

Regards,

Mathieu Peyréga

LuboBonchev, 2015-04-27:
Hi Matioupi,

I expect to have the next MoPi in Summer. We are delaying to the planed schedule, but testing of different prototypes takes long time - charge the batteries, discharge the batteries, repeat this many time in different scenarios and with different batteries .... Hope you understand.

Will send you the full spec when have it updated. Some details got changed during development. 

Regarding the features:
   --  You forgot to wish embedded charger, but you will have it :-) Programmable, multi-chemistry. 
   --  Yes, 3 power inputs: 2 as now plus +5V micro-USB
   --  4A are toooooo much. According to the published by the foundation requirements the power should secure 2A including the power for USB devices. Pleaae note that Pi2 and Pi+ versions have lower consumption comparable to the first Pi versions. My understanding does not match also 4A power consumption and portable battery packs. 
   --  Will add PCB holes (or small connector) for external LEDs and power switch. 

Do you like to be a bug shooter for the next MoPi? If so, please send me your address and I will send you one prototype when is ready. 

Thanks,
Lubo Bonchev

stoduk, 2015-04-27:
As people are offering suggestions..

Have you guys considered adding an RTC in to the mix?  Given MoPi exists to
allow devices to run off batteries, I'm surprised more devices aren't after
reducing their power usage (which means ditching network connection and so
NTP, and gaining RTC if awareness of time is needed).

Of course, if the RTC could be accessed by both the RPi and MoPi then this
adds the possibility of adding more time based operations for MoPi in to
the mix (eg. waking up at some distant future time, rather than having to
specify some delay from when the RPi is next shut down).  I don't know if
there is some issue with having two masters for the RTC if it talked i2c -
wikipedia suggests i2c is multi-master/multi-slave so it may just work.

Thoughts?

LuboBonchev, 2015-04-28:
Thanks Anthony!
Yes, RTC is included in the new version. The microcontroller that is driving MoPi has RTC in-build. The firmware is making it compatible to the supported by the Pi drivers.

stoduk, 2015-04-28:
Great, that is one less extra component to worry about for future
projects.  Given how quickly you agreed, makes me think I should be asking
for more :)

LuboBonchev, 2015-04-28:
That because the RTC was added to the spec a year ago :-)  Nevertheless we are open for discussons of any feature. At least the prototypes are done :-)