hamishcunningham/pi-tronics #40: configuration Gui will not change to reflect my battery/batteries

david1944, 2015-04-22:
Every time I try to change an element in the Gui it doesn't ha[[en.  THree scenarios:- 1) Two battery holders AAA non rechargeable.  2) Two PP3 non rechargeable, and 3) one rechargeable PP3 180mAh NiMH (favoured for two projects I'm working on .. a robot vehicle to carry maplin arm, and a static crane made of balsa with a duty cycle reduction to make the winch motor and gears run slowly.  All is ready for battery power, but it doesnlt work.  My scenario - Attached MoPi OK,. B+ powered up via mains and monitored on Windows 7 PC via command line and xrdp. Call cli and write sudo mopi.  Config Gui comes up.  The only key that seems to work is the 'tab' and that does nothing except move the screens on.  At the end of config the cli has not changed,  Tried starting  by using cli and rpi does not recognise /usr/main/mopicli etc.for rechargeable battery (-wc1 1 9600 72  6335 5740)  .  Help please.

hamishcunningham, 2015-04-22:
hi David,

Can we break this down a bit? First problem is that the "mopi" UI should work :-)

The UI uses the same tool as the RPi's own configuration utility, so if ours doesn't work it suggests that either we're not connected via a usable terminal interface or that the simbamond software installation is corrupted. I'd say the former is most likely.

Things to try: 

- run raspi-config over the same terminal and see if that works
- connect a monitor and keyboard and log in to the console
- reinstall simbamond: sudo apt-get update; sudo apt-get purge simbamond; sudo apt-get install simbamond

Good luck, let me know how you get on!

Best

Hamish

david1944, 2015-04-23:
Date: Wed, 22 Apr 2015 01:12:26 -0700

Iv'e done all that and more.  Ilve searched high and low on Raspi to find the file that the config is saved to.  Short answer ;Canlt find it any where in Raspi.  If this is indeed the case then I cannot amend a file which is not on my microsdh card.  Purged and re-installed.  Updated and upgraded.  Changed batterrie to ones with very nearly 9 volts, (they are ex fire alarms that have had little or no use)OK so have I had ant sucess. Yes measured. 2 batteries reading 8.7 volts lit mopi like blackpool illuminations - BUT - only got as far as the login request.2nd trial was with an Adafruit tft screen.  Baclkight shiny and new, loaded perfectly, logged on and then - It shut down?  What was different from yesterday's failures is simple.  I tinned all four connections and they now fit into the terminal block.  The two black double-up, I think, is less secure and more fiddly that having four terminals.  I'm considering joining them outside Pi and only having one wire inserted into the mid terminal.The whole problem comes back down to amending config, which Ilm sorry to say still does not work.  I've tried it through xrdp in windows; Ilve tried it with keyboard and minitor attached to raspi. I'm now going to try Putty but I donlt see it making any difference. So here goes.......nochange.  When I put up status report there is a screen behind it which looks as though amndment has worked, but normal shows your default.
Ireally don't know where to go fom here. Ilve bought a new charger and will soon have 8 AA rechargeables to try.  While Ilm waiting I'll make a balso wood box to collect them in with external connection wired to one plu and one minus..
Catch you later. Have a good day.

hamishcunningham, 2015-04-23:
the config is saved into simbamond.default -- does that help?
did you get the UI to work?
if so what did it report when you ran the configuration steps?
tnx
h

david1944, 2015-04-23:
Date: Wed, 22 Apr 2015 01:12:26 -0700

Hello Hamish,It's bad news I'm afraid.  I cannoitget the config UI to work.  I tried all you said and it half worked and then collapsed.  In the final death throws I decided that my new micro 16gb could be sacrificed in the cause. I loaded Noobs, did my config entries, changed my password and then updated, re-booted and upgraded, rebooted again and install simbamond, rebooted once more and loaded mopi.  Got the config UI and attemted to alter it.  NBG -was not having it.  Found mopicli and had a good look but iy didnlt lead me anywhere.  I cannot see a way around it. I tried putty  and no change. I haven't loaded xrdp yet.  I canlt find 8 AAA rechargeables onmly 6, and that cannot work with what seems to be a concreted configuration.. I will keep trying anything you suggest.Best wishes,David.

hamishcunningham, 2015-04-23:
What happens with raspi-config?
When you say "doesn't work" what's the specific behaviour?

david1944, 2015-04-23:
> Raspi-config works perfectly both on the old 'card' and on the new.  On the old one I specifically changed my password to make sure it worked.  On the new one I've done all the things it is usual to do.Changed password, enabled camera, and the advanced options including i2C. Specifics follow:-
> sudo mopi return
> up comes the configuration UI.
> Change the power supplies no1 return,
> Option 2, non-rechargeable enter (it didn't feel as though it had changed)
> Taken sraight to next screen choose 6 cells for PP3 return (it didn't change the position of the marker from 8 cells)
> taken to next screen, chemistry alkaline return (same comment-it didn't change}
> taken to final screen where the results are shown and you're asked to confirm if they are correct.  Well they were incorrect.  They show the same basic data that you put into your command line equation. Important points being NiMh should be alkaline,  and the four calculations 11200,10000, 8800, and 8000.
> This means that the config hasn't been (1) amended and (2) that the calculations would not allow PP3's to work without going quickly into criticak and stop.
> so to finish off, press yes and then OK and back to main screen to do battery no2
> .  Same results
> What would I expect to see on each config screen?  A question to confirm the change and then (perhaps) a next button, to take you to the next screen.
>
> I tried the command line instruction and was replied with - something like- bash no such command.  I wondered if I should be editing a line in mopicli, but no, /usr/main/mopicli -wc1 1 9600 7200 6336 5760 only produced a new file, so I pressed cirl x and returned to $prompt.
>
> As to my remote monitor, keyboard and mouse. I use a dos command line with mstsc.exe and the ip address of raspi.  On raspi all I have to do is to install xrdp.  The service runs sesman and links directly into LXDE Gui unlike putty and ssh which only links to the raspberry's command line.
>
> David

Date: Thu, 23 Apr 2015 14:18:28 -0700

hamishcunningham, 2015-04-24:
"it didn't change the position of the marker" -- how are you trying to
control the UI? arrow keys plus tab and space and so on?
tnx
h

david1944, 2015-04-24:
> ?space?  That was the missing element.  I have now amended the UI but it still goesn't boot.
> Might be a battery problem so will try with another pair, another pair until I get success.
>  I'll let you know what happens and many thanks. Hmm SPACE. never thought of that.
> David.

Date: Fri, 24 Apr 2015 00:13:07 -0700

https://pi.gate.ac.uk    https://hamish.gate.ac.uk   https://gate.ac.uk

hamishcunningham, 2015-04-24:
:-)

the penalty for being compatible with very old tools like terminal
emulators!

good luck

best
h

david1944, 2015-04-24:
> Now working although not quite enough power to get past login and password but I'm working on that.  What's the absolute maximum power that the board will take?
> Thanks for your help and support.
> David Lake.

Date: Fri, 24 Apr 2015 00:13:07 -0700

https://pi.gate.ac.uk    https://hamish.gate.ac.uk   https://gate.ac.uk

hamishcunningham, 2015-04-24:
mopi can supply around 1A to the Pi, with input voltage between 6.2 and 20V
the board itself doesn't draw much power in comparison to the Pi and its
peripherals
wrt batteries anything from a pp3 upwards will work, but you won't get much
life with less than e.g. 6 AAs (and note that the capacity of these varies
by almost an order of magnitude)
more details e.g. here https://pi.gate.ac.uk/pages/mopi.html#section-6.2.
hth
h

closed 2015-05-21 by hamishcunningham

david1944, 2015-05-26:
Hello Hamish,
I don't know whether you can help with this problem but I've puzzled over it for a long time (it seems long anyway).
Ok install mopi on B= and run from batteries :-
result:- sh: can't access tty: job control turned off. This comes up after rainbow screen.  The recovery keys don't work.  Only a green light shows on B=.  type exit in sh and a statement comes up about 'killing kernal' and process of booting fails.  Tried booting from mopi and yellow lights for both batteries and a green light . Green turns to red but mopoi doien't always turn off. When this happened first time I panicked and disconnected mopi which shutdown B+.  I've bought another B+ but darenot use mopi with it in case same thing happens.  All new sdhc cards( formatted, noob, asbian setup all work.  One configured in new B+ and put into old B+ gives job error even when cmdline.txt is changed to avoid safe mode.  Shorting has no effect.
What can I try?
David.

hamishcunningham, 2015-06-05:
this post suggests a new firmware might help? http://raspberrypi.stackexchange.com/questions/19561/cant-access-tty-job-control-turned-off