hamishcunningham/pi-tronics #41: Power on at specific time?

wayner9, 2015-04-25:
I am using my Mopi in a solar powered configuration.  For my application it might be better to power on at a specific time (ie 0800) rather than X seconds after powering off.  That way I could ensure that it is daylight.  Would this be possible with mopi?

stoduk, 2015-04-26:
Hi,

Unfortuately you can't do what you want directly, but you can work around
it depending on your setup.  Longer answer below, as I had to solve this
for a project I worked on.

The power on timer only starts ticking when the MoPi has turned off the
RPi.  So you have to set this at the point where you know when the MoPi is
going to turn the RPi power off.  You can know that by always setting the
MoPi's power off timer at the same time.  So set both at once and with some
simple enough sums you can emulate what you want.

If you also want to figure out sunrise/sunset you can do that, there are
libraries out there.  My project has this, to try to only take photos
during daylight hours.

To have a knowledge of real time (rather than just elapsed time) your
device will need to either have network access (which increases power
usage) and NTP, or a separate RTC.  I went with the latter combined with a
Model A - to try to get maximum battery life possible.

I did all this for a time lapse camera project that fell foul of the Kid
Sister Law (if she can break something, she will).  When I get the time
lapse back I will hopefully be able to bring it back to life and document
it all more.

For now though the source is here, which is all pretty self-explanatory:
https://github.com/stoduk/tempus_fugit
I also put it on hackaday, but that hasn't been updated much thanks to the
problems I mentioned :)
https://hackaday.io/project/3288-long-term-time-lapse

wayner9, 2015-04-26:
Can you run a python script at the time that Mopi is shutting down the Pi?  If that is the case then couldn't you just set the Power on seconds offset from the current time to give you the power on time that you want?  

For example if you want the Pi to come on at 8am and the Pi is shutting down at 8pm then you set the power on offset in seconds to 3600*12 or 43200?  You would need a little bit of logic to avoid negative offsets but that is easy peasy.

stoduk, 2015-04-26:
Sure, setting the power on timer as the MoPi is shutting down the RPi is
all that is required - whether you do that by setting the power off timer
at the same time as I did, or running a script at shutdown.

For me the former was simpler as my RPi was basically waking up, doing some
work, then powering down straight away - so setting both timers at once was
simplest.

One side note: I forget what the range of value for the power on timer was,
but it wasn't huge.  Certainly you couldn't set it to wake up a long way in
the future.

Hamish and co can comment on what the max power on timer value is.

wayner9, 2015-04-26:
When the mood shuts down the Pi due to dropping voltage what method does it use - a shutdown?  If so does that mean that I should be putting a shutdown script in /etc/rc6.d?

stoduk, 2015-04-26:
https://github.com/hamishcunningham/pi-tronics/blob/master/simbamon/simbamon#L35

Looks like it runs "halt" command.

I've not played with rc6.d/init.d scripts in a while, so can't really help
there.

[if you didn't know, I'm just a user, my project just had some of these
problems solved for me]

hamishcunningham, 2015-04-26:
Thanks Anthony!
h

wayner9, 2015-05-01:
Hamish - any advice on how to power on at a specific time?  Have you ever done these types of scripts before?

hamishcunningham, 2015-05-21:
I think Anthony answered this one, so closing the issue.

closed 2015-05-21 by hamishcunningham