hamishcunningham/pi-tronics #42: In-Car Mopi circuit not working as expected

r08shaw, 2015-04-29:
As discussed previous on Disqus:

I've think I've got an issue with the circuit. Can I just check with you how the circuit should function?

I was testing WITHOUT an RPi connected as I wanted to make sure everything was working before risking a deep fried pi. On connecting the 12v power source (8xAA) I was expecting some of the relays to trigger, but this is not the case. I've checked for dry joints and shorts and everything is fine. The relays are ok too as I tried directly connecting the battery and all could be made to switch.

I've been using a meter as a continuity tester and it seems that the 5v and 0v lines, that connect to relay 3/diode 3 (on your circuit diagram), are shorting. I've got a 2 pin header on these connections to wire up to the Pi, but they are already making a circuit. Even without the header

HELP!!! Should I be testing with the Pi, is that the issue?

Eagle Schematic
[image]

Eagle Board layout
[image]

Latest Eagle files here: https://www.dropbox.com/s/qilyx6xzh15kvmn/InCarMoPi-eagle%20files.zip?dl=0

hamishcunningham, 2015-05-03:
hi Rob

I can't see anything immediately wrong with your schematic. Without seeing where you're placing your multimeter probes it is hard to say why you're seeing the behaviour you are... Perhaps you have a diode reversed? These conduct in one direction and not in the other -- their function in this circuit is a kind of "debouncing" when the relay opens to prevent surges. If they were the wrong way around then they would instead short the relay coils and hence prevent them triggering? 

Or possibly you have a defective part, with the same effect.

Sorry I can't be of more help -- I really would advise trying to find someone in your region who you can show the beast to in person?

Good luck! Best,

Hamish

r08shaw, 2015-05-03:
If it helps, I'm placing my probes on V1 and V2 of my circuit. These should go to 5v and 0v on the Pi. On your circuit they should go to pins 2 and 6 on the GPIO. They are shorting at the moment.

Will double check the diode placement. They are all pointing in the same direction, so if they are wrong, then they would all need flipping.

closed 2015-10-16 by hamishcunningham