hamishcunningham/pi-tronics #43: How does the Mopi decide which Power source to use

wayner9, 2015-05-08:
If you have two sources connected how does the MoPi pick whcih source to use - does it always use the higher voltage source?

Is it possible to have the MoPi always favour one source, even if it has a lower voltage?  I am using a MoPi with batteries and a solar cell.  As long as the solar cell is generating sufficient voltage I would like the MoPi to use it rather than the batteries.  Is there any way to do that via software?

hamishcunningham, 2015-05-14:
it always uses the higher voltage source
hth
h

closed 2015-05-14 by hamishcunningham

jeberhardt, 2015-08-17:
Does this mean if there are two sets of 8xAA that it will swap between the two sets, so that both of them arrive at 0 around the same time?

It would be preferred if it drained all the power from the first set, leaving the second set with power.  This would allow the removal of the first set to be recharged while the second set was being used.

Can you please clarify this?  Thanks.

hamishcunningham, 2015-08-17:
yes, we realised this a little too late :-(
the next version remedies, IIRR
best, h

jeberhardt, 2015-08-18:
Would this be a firmware update on the current hardware, or would this require new hardware?