hamishcunningham/pi-tronics #47: Mopi and B+

david1944, 2015-07-30:
I now have two (2) B+ boards that demonstrate the same problems.  The only, repeat only, board that has been attached to both is MOPI.  The fault generated by using Mopi within its rules are a non-functioning set of usb ports and lan port.  The only time during my trials of mopi have either been attached to mains supply was to fill out the configuration on command line. Mopi worked with the first board intermittently and with the second board only once and then although it booted Raspberry it failed to get as far as login; it loaded simbamon and then reset itself and stopped at the rainbow screen.  There was no message to show what had cause it to stop; hey, and guess what I can't log in or connect to the internet brcause the usb/lan ports are not working.  I have just finished trying to get a result using a serial connection, but, although I can get one to boot to login I cannot use my keyboard or that of the connecting Debian PC. Passive connection through putty doesn't even give me a login or password entry.  Has anyone suffered the same fate?  Hamish, if you or your team have any ideas for a solution then please let me know quickly.

hamishcunningham, 2015-07-30:
hi David

The only thing that I can think of is that you're connecting batteries that have insufficient charge to persuade MoPi to power the Pi.

There are a range of problems that you might be hitting with the Pi itself -- have you tried a new OS, for example, in relation to the USB problems? At some point the older versions of Raspbian suffered from this problem with the B+ IIRR. Perhaps you might remove MoPi from the equation and pursue these problems separately on the Pi forum? 

It seems quite unlikely that you're the first to hit some new fundamental problem -- there are around 1000 boards out there which are working nicely :-)

So I would recommend:

- removing MoPi and powering from the mains
- reflashing your SD card with a new OS
- when you're back up and running trying MoPi again, but paying particular attention
  to config and battery charge and the like

Hope this helps, best,

Hamish

david1944, 2015-07-30:
Hello again Hamish,Thanks for your prompt response, but I have tried all of your suggestions before I reported the issue.  With the USB and LAN ports out of action It matters not whether the right noobs is or has been applied.  If you can't attach a keyboard or mouse then you cannot use a re-flashed card at all.  Mopi is the only link that I can see clearly   As a last resort I thought that using a PC to boot raspi from the serial port was worth a try, but the same thing happened - without a keyboard I cannot instruct (interact with) raspi. to use minicom or even to use recovery shift - but then the LAN's absence would stop the reloading of raspbian. I did wonder if, with your knowledge and academic status you could suggest a more unique  and bespoke answer; a script that could be fed into the recovery command line perhaps. because my windows PC can open the sdhc card and with notepad++ can edit the file. I have a PC that runs debian ; it has python.  Its HDD is 160gb and only a couple of weeks old. Can that be used?  I am in your hands so to speak.  If you can help i would appreciate it.Hope to hear from you soon,All the best,David. 

Date: Thu, 30 Jul 2015 08:19:35 -0700

hamishcunningham, 2015-07-30:
It is worth reflashing anyhow as that may fix thr usb prob， which is prob
software not hardware?

david1944, 2015-08-01:
> Dear Hamish, Sorry to put you to the trouble of having to read fully my statement and then to give a coherent and considered response.  There is a simplicity about my rasp[i problem which you have failed to grasp.  That is:- Without USB and LAN ports you cannot a) use recovery command line, and, b) you cannot connect to the interrnet.  I still feel that the problem is a hardware one.  So how do I find a workaround.  More of that later.  I have 12  x 9v0 batteries.  Before I contacted you I did the following exercise,  Each battery had a sticker with a letter A<B<C< etc and each was measured by my multimeter before being used and that value written on the label.  I then connected pairs of batteries to Mopi and noted the results.  The voltages ranged from 9.74 to 7.35.  One pair showed a result; batteries G an I - with
>  a combined voltage of 16.31 volts showed a steady blue on mopi and Pi had a steady red and a flickering green.  It did not boot pi. Took G off and all dead. Took mopi off and re-loaded the batteries and this time all dead. Measured the output an it haDn't varied greatly.  Took I of and replaced it with D and the combined volage was 17.14 ;result red on mopi flashed but didn't light pi. Then tried G&I again and dead.hi David.
> And you say pay particular attention to batteries.   Yesterday I recharged four NiMH batteries and tried them without a result.  I then took Mopi off and booted via mains,  Mopicli still not in evidence.  No login - no usb keyboard.  Bought a new usb keyboard and usb mouse; no difference no usb.  I put I then changed the card for a boot into X gui.  No mouse control.  Then put adafruit resistive touchscreen tft on board to allow me to re [... cut ...]
> I have tried to use my RS232 module but I cannot get an internet ip[ or a reply from Raspi - no keyboard.  I can load the sdhc into Debian but the files I need to change are not editable without raspi's root login.  It is this area that I need help with.  I simply believe if the fault is software then software can correct it.  I have six sdhc's ready and one which I noo bed some days ago which I cannot boot. So you see my logic is reasoned.  Please read all of my messages before responding.

Date: Thu, 30 Jul 2015 08:19:35 -0700

hamishcunningham, 2015-08-01:
I have had a USB malfunction on a Pi before -- I know what it is like :-)

In my case, taking out the SD card, putting it in another machine and
reflashing a newer OS cured the problem.
Have you tried that?

If not, why are you convinced that the USB problem is a hardware fault?
Thanks, best, H

closed 2015-10-16 by hamishcunningham