hamishcunningham/pi-tronics #53: Calculating power consumption

jeberhardt, 2015-08-18:
I am trying to calculate how long my batteries should last, and am getting confused with voltage vs. amp hours, or milliamp hours.  There are a few related questions, which I'm hoping you can help me out with.

I have 2 sets of 8 AA Eneloop Pro batteries, for 9.6 volts and 20,400 mah on each pack of batteries.  

I have a Charger Doctor USB multimeter that shows me the A+ RPi with it's USB peripherals is drawing  about 300 mah (when plugged into a USB power source).  That would lead me to believe that the RPi might last about 68 hours (20400 / 300) on a single pack (which I find highly doubtful).

How does the voltage play a role in calculating how many battery mah get used in something that normally draws 300 mah when plugged into a 5V wall plug.  Is there a loss of available mah when powering this way?

How does connecting the batteries in series vs. parallel affect these calculations for the Mopi?

Sorry if this has been asked already - I took a look through the archives but couldn't find an answer.  Great product, I expect I'll have more questions as I use it more.

stoduk, 2015-08-18:
[I'm just a user, commenting because I'm nosy and this is more interesting that the email I need to write :) ]

Ampere-hours measures charge - not quite the same as useful power you can get out, but similar.  Cells in series only increases the voltage, not the charge capacity.  Cells in parallel would increase the charge capacity, but not the voltage.  Found the page below which might help explain that more..

So assuming you have 2000mAh cells, then 8 in series would give you 9.6V and 2000mAh.  So a rough expected use of 2000/300, which is 6.66 hours.

See also the MoPi page below which has done some tests of usage in certain situations - remember that there are various factors that effects how long something that can run off batteries, including age of batteries, how well they've been looked after, how many recharge cycles they've gone through, temperature, etc. 

http://www.zbattery.com/Connecting-Batteries-in-Series-or-Parallel
https://pi.gate.ac.uk/pages/mopi.html#section-6

HTH,
Anthony

jeberhardt, 2015-08-19:
I had 2x8AA packs (for a total of 16 batteries).  I was able to have the RPi straight with no interruptions for almost exactly 28 hours.  I know that batteries can be quite different because of the circumstances you've mentioned, but it still doesn't help me understand how I can possibly even come up with a rough guess as to how long the batteries should last.

Thanks for any additional information.

stoduk, 2015-08-19:
Was the RPi drawing 300mAh when you saw 28 hour usage?  Were the 2x8AA packs each wired to MoPi as a separate power source (ie. effectively in parallel)?

Assuming that then you've got 20400/8=2550mAh cells, with two batteries of 8 cells each giving a total capacity of 5100mAh.  At 300mA that would last 17 hours.  At 100mA that would last 51 hours.  At 182mA that would last the 28 hours you saw.  Obviously these are theoretical numbers, I'd expect real world to be a bit lower (due to manufacturers lying about capacity, cell capacity declining with use, imperfect efficiency in any power conversion).

Apparently the idle usage of A+ is 100mA (see link below) - so perhaps your RPi load was lower in your 28 hour test (ie. CPU load + connected devices), than when you had the meter attached.

One other factor that might be significant is efficiency of the MoPi at different voltages - I don't know if that might have any effect here.  8xAA is the most common setup for Hamish and co, I don't know if that was chosen for efficiency reasons or not.

Hamish and co. may want to comment more, but largely I think measured numbers beat theoretical numbers any time :) If you want to see worse case numbers then I'd run similar tests with high CPU load.

HTH,
Anthon

http://raspi.tv/2014/raspberry-pi-a-how-much-power-does-it-need

hamishcunningham, 2015-08-19:
Anthony's spot-on: estimating battery life is a very inexact science, and
measurements are the best way to go. The 28 hours sounds in the right
ballpark with a mixed load on an A+ and decent batteries. Incidentally I
recently bought one of these chargers (with individual level indicators and
independent charging) and have saved myself *loads* of time!
http://www.amazon.co.uk/gp/product/B00BO5A5ZA

HTH, Hamish

closed 2015-10-16 by hamishcunningham