hamishcunningham/pi-tronics #55: LiFePO4 Batteries

jeberhardt, 2015-08-19:
Any comments or recommendations for LiFePO4 batteries?  I would like to find a solution that will power a RPi A+ for 48+ hours (8 hours a day for 5 days, plus a little extra).

Currently I have 2 x 8AA 2550 mAh Eneloop Pro NiMh, which powers the RPi for about 28 hours.   I need to find something that will give me that extra 12-20 hours.

I'm not so worried about the cost of the batteries, but I am concerned about the size, weight, and logistics of recharging 16 individual batteries.

stoduk, 2015-08-19:
In my experience lithium batteries tend to be over sold even more than NiMh, so bear that in mind when looking at capacities..

If runtime is your main consideration then a (deep-cycle) lead acid battery might be the way forward.  A quick look at Halfords suggests a basic one is 40Ah, though I've seen mention of 120Ah batteries.  Charging is obviously simple, just connect a regular car battery charger.

[Do you need the RPi to be running all the time?  If not you could do what I did for my time lapse project - only turn on periodically and then only long enough to do the work that is needed]

hamishcunningham, 2015-08-19:
Lead acid is definitely cheapest, and available in lots of sizes from e.g.
Maplin.
Or add more AAs -- I think you can just about connect 16 AA NiMH to each
MoPi supply.

closed 2015-10-16 by hamishcunningham