hamishcunningham/pi-tronics #56: Li-Ion 18650 Configuration

jeberhardt, 2015-08-22:
I've got some 18650 batteries that I'm trying to connect to the Mopi, and it's telling me I've got low voltage - I think the configuration is wrong.

I've got 2S5P connected to one power supply - that's 5 cells connected in parallel, which are then connected in series to another set of 5 in parallel.  I'm pretty sure I've got them hooked up correctly.  ;)

My multimeter reads the voltage not under load as 8.32V.  However, I suspect the nominal voltage will be around 7.4V, since these batteries are 3.7V.

In the config setup, I choose (1)rechargeable batteries, (2)2 cells, (5) Lithium Ion.  It then said the minimum safe was 8.0V, the low charge was 8.8V, the good charge was 10.00V, and the full charge was 11.2V.

I would have thought the minimum safe charge was 5.0V, the low charge was 6.0V, the good charge was 7.4V, and the full charge was 8.2V (roughly).  

Is the configuration tool wrong, or is my understanding of the voltage wrong (likely the later).  Should I manually override the settings?  Thanks for any input you may have.  I'm a little baffled by this.

[image]

stoduk, 2015-08-23:
Hopefully helpful: depending on which type of lithium cell you selected,
you should get the values below.  This is using the calculations the config
setup should be using.

Might be worth rerunning the config and double checking you've selected
what you meant to - I remember the menus are a bit odd (you don't just
scroll down to the option you want, you have to highlight it - I forget the
keys, but space bar rings a bell.  It will then have an asterix by the
selected option IIRC).  This seems like the only way I can think that this
could be user error..

If there is a genuine problem here then I'll leave it with the MoPi folks
to comment :)

bash $ test_dump 1 2 5

"Rechargeable batteries"

"Two cells (e.g. double 3.6V LiPo)"

"Lithium Ion"

floor 6.0/low 6.6/good 7.000/full 7.4

bash $ test_dump 1 2 6

"Rechargeable batteries"

"Two cells (e.g. double 3.6V LiPo)"

"LiPo"

floor 6.0/low 6.6/good 7.500/full 8.4

bash $

test_dump() {

supply_type $1

num_cells $2

bat_chemistry $3

calculate_config $1 $2 $3

echo "floor $BAT_FLOOR/low $BAT_LOW/good $BAT_GOOD/full $BAT_FULL"

}

jeberhardt, 2015-08-25:
Good news!  I was able to re-run the configuration (pressing the space bar to select the item), and got the following values:
Maximum voltage: 7400 v
Good voltage: 7000 v
Low voltage: 6600v
Critical voltage: 6000v

I got a message indicating that the effective critical voltage would be 6200v, and that's fine with me.  I'm really not sure why the configuration didn't work the first time around, as I re-did it several times and it didn't take.  It's working now though, so that's good.

closed 2015-08-25 by jeberhardt

hamishcunningham, 2015-08-25:
I should add this to the FAQ -- few people are used to the old curses (sic)
interface style these days!
glad you got it to work
note that 6.2V is the absolute minimum for MoPi's regulator, so you have
little headroom in that setup
hf
h