hamishcunningham/pi-tronics #57: Loss in voltage

jeberhardt, 2015-08-25:
There seems to be a difference between the voltage of the batteries themselves and the output of mopicli.

If I use mopicli -v, it'll show 7100 for example.  If I take the leads from my multimeter and apply them to the wires connecting the batteries, it'll show 7.40 (the equivalent of 7400).  I'm wondering what is accounting for the difference in voltage, and if there's a way I can reduce that difference, ensuring I'm getting everything possible out of the batteries.  Thank.

hamishcunningham, 2015-08-25:
Batteries under load vs. not?

guruthree, 2015-08-26:
A small voltage loss is unavoidable across the Schottky diode we use to balance the two input sources. This sounds like what you're measuring!

jeberhardt, 2015-08-26:
That seems to be the likely source of the voltage loss.  Thanks for the explanation.

closed 2015-08-26 by jeberhardt