hamishcunningham/pi-tronics #58: Low battery warning too soon?

jeberhardt, 2015-08-26:
I saw that the amber light to my battery pack was blinking, which I gather is an indication that the voltage is too low.  Looking at the battery voltage, I see the following information (see attached screen shot).  

It shows that the low voltage is set for 6600 mV, whereas the current voltage is 6786 mV.  Can you explain to me why I'm getting the warning that the voltage is too low?  

Thanks!

[image]

guruthree, 2015-08-26:
The LED flashing for low voltage triggers the first time the battery drops below the low voltage level, and then stays flashing, even if the voltage rises again, until the battery has been replaced. What's probably happened then is a peak in load from the CPU or a device has caused the voltage level to temporarily drop, after which it's risen back up. +/- 200 mV is a fairly standard voltage fluctuation.

jeberhardt, 2015-08-27:
Good to know.  That sounds precisely like what would have happened.

closed 2015-10-16 by hamishcunningham