hamishcunningham/pi-tronics #66: Charging Circuit with MoPI

ginzer, 2015-12-04:
Hey  I want to have a Raspberry working with a rechargable Battery and a power supply. When I plug in the power supply battery should be charged and when the battery is full charging should stop and the raspi should work with the power supply. When I plug off the ower supply Raspi should work with the Battery.  Something like that. But i dont really get it. is U4 conected to the MoPi supply 2 or to the USB supply from the PI?
[image]

The other one i found here shoul work......
[image]
or am I wrong? I just dont need the amplifier and stuff....

Thanks for any help.

hamishcunningham, 2015-12-04:
howdy

looking back at the first (UPS) circuit (where U4 is a bog standard 5V supply connected to the micro USB on the Pi) it relies on the 5V supply overpowering the supply from MoPi... 

but yes, the second circuit sounds more like what you need; just delete U3 (and, if you like, specify a lower current 12V supply, or 9V supply, for U2).

hth, best,
h

ginzer, 2015-12-04:
thanks. and the system wont shut down because the relais is to slow?

hamishcunningham, 2015-12-04:
not in similar circuits I've built before, but YMMV :-)

a mosfet would do the job just as well, though you don't get the cute
clicking noise :-)

hf
h

ginzer, 2015-12-04:
thanks a lot.

blackstarek, 2016-01-03:
Howdy, @ginzer , any progress? I just bought mopi and i miss this function, how did you succeed?

ginzer, 2016-01-07:
hey....no no progress. I just decided to use a Powerbank wit pass-through function.

blackstarek, 2016-01-12:
@ginzer may i ask you which powerbank did you use and how did you connect it?

ginzer, 2016-01-18:
its a limefuel. I just connected it to the power supply of the raspy

ginzer, 2016-01-22:
but to be honest the i would prefer the solution with the mopi.
@hamishcunningham is there any current limitation for the mopi? My Raspberry needs arround 2A.

hamishcunningham, 2016-01-22:
2A is the max output :-)
h

ginzer, 2016-01-22:
ok thanks :-). perfect

so I could make it like that talking about shematic 2):

instead of the 220V supply I use a 5V supply which gets to the Lipo charger and to the Mopi.... and if I take away the supply the battery will supply the Mopi (and the Raspi.....)

hamishcunningham, 2016-01-22:
MoPi requires a 6.2V min supply voltage, and it is typically easier to
arrange things with a higher voltage, e.g. 12V

hth

h

ginzer, 2016-01-22:
ok. so if I use a 12V supply it should work.

closed 2016-12-19 by hamishcunningham