hamishcunningham/pi-tronics #68: MoPi is dead?

gavioli, 2015-12-17:
Good evening
I am your Kickstarter supporter, my card is numbered with 0422.
I never used the card in any project and now I was going to use it with a raspberry and touch screen 7 ", I realized that the card does not work.
After installing the software and inserted the MOPI on Raspberry, no LED It turns on and any battery that I try to connect will overheat and discharged.
I wanted to ask how you can tell if the cause is in raspberry or if the MOPI is dead. If the device that don't  working is MOPI, you can get information on possibile repair?
I'm available to provide more information

Thanks for the attention.

[image]
[image]

hamishcunningham, 2015-12-17:
hi

Sorry to hear that. All units were thoroughly tested before dispatch (which is over a year ago now) -- I wonder if the unit has been physically damaged during that time?

Also, how did you connect the batteries, and is the Pi you connected to operational?

Thanks, best

Hamish

gavioli, 2015-12-18:
Hi, again thanks for your attention.
I am sure that the card was sent running.
Probably the malfunction was caused by my incorrect connection or whatever.
I disturbavo to ask if I could have the directions on how to try to fix the card, failing to provide the complete circuit diagram.
Comparing the only scheme available online (I hope is that referring to the card in question) with the PCB, I have noticed that the capacitors to tantalum (10 microF) in input to the batteries, are placed not after the diodes, but before and my reversal polarity of the batteries may have been damaged them.
Just after boot, I could check the following:

login as: pi
pi@192.168.0.221's password:
Last login: Mon Dec 14 18:28:41 2015 from pc2230.local
pi@raspberryMotion ~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberryMotion ~ $ date
Mon Dec 14 19:34:39 CET 2015
pi@raspberryMotion ~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberryMotion ~ $ sudo mopicli -v -v1 -v2
mopicli. I2C bus input/output error on read word. Check bus? Check connection?
pi@raspberryMotion ~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberryMotion ~ $ date
Mon Dec 14 19:35:07 CET 2015
pi@raspberryMotion ~ $

and the main LED never come on.
Perhaps with this information you can give some indication as to which component may still be damaged.
I apologize again for all this trouble, I wish you good day.

hamishcunningham, 2015-12-18:
thanks for the details -- you can see how a MoPi fits on a Pi B+, for
example, in this video: https://pi.gate.ac.uk/posts/2014/08/02/b-plus-again/

the batteries connect with a common ground underneath as detailed here:
https://pi.gate.ac.uk/pages/images/mopi/quickstart/schematic6.png

is that how you've got it connected?

best

h

hamishcunningham, 2016-01-22:
did you get any further with this?
tnx
h

gavioli, 2016-01-29:
Good morning, thanks for your interest!
As described in the previous message, the problem that does not allow me to use the card, is that after a few minutes from the boot of raspberry, querying the bus with the command:
"sudo -v mopicli -v1 -v2
mopicli. I2C bus input / output error on read word. Check bus? Check connection? "
it seems that the card is not detected.
The batteries and the LED are working properly, but only until the card does not lose the connection to the bus.
We report querying the bus:
pi@raspberryMotion ~ $ sudo i2cdetect -y 1
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- 0b -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
pi@raspberryMotion ~ $ sudo mopicli -v -v1 -v2
mopicli. I2C bus input/output error on read word. Check bus? Check connection?
pi@raspberryMotion ~ $ sudo i2cdetect -y 1
0 1 2 3 4 5 6 7 8 9 a b c d e f
00: -- -- -- -- -- -- -- -- -- -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --
I tried to change as the raspberry and the raspbian version, but nothing changed.
Have you any suggestions?
Have a nice day
Flavio

hamishcunningham, 2016-01-31:
ok, if you email me via https://hamish.gate.ac.uk/pages/contact.html your
address I'll send you a replacement?

please ensure that you're not connecting too high a voltage or connecting
power on the wrong pins or the like

hth, h

gavioli, 2016-02-11:
Good morning,
I just received your package, thank you very much, it was really very kind and also very timely.
I connected the new MOPI and am very pleased to tell you that in works great!
I still have to finish configuring the different battery packs, but as you can see the checks are ok.

pi @ raspberryTEST ~ $ sudo -v mopicli -v1 -v2
Current source voltage: 11466
Source # 1 voltage: 11466
Source # 2 voltage: 6864
pi @ raspberryTEST ~ $ sudo i2cdetect -y 1 0 1 2 3 4 5 6 7 8 9 a b c d e f
00: - - - - - - - - 0b - - - -
10: - - - - - - - - - - - - - - - -
20: - - - - - - - - - - - - - - - -
30: - - - - - - - - - - - - - - - -
40: - - - - - - - - - - - - - - - -
50: - - - - - - - - - - - - - - - -
60: - - - - - - - - - - - - - - - -
70: - - - - - - - -

more @ ~ $ raspberryTEST dates
Fri 12 February, 2016, 00:12:23, CET

pi @ raspberryTEST ~ $ sudo -v mopicli -v1 -v2
Current source voltage: 11466
Source # 1 voltage: 11466
Source # 2 voltage: 6864

pi @ raspberryTEST ~ $ sudo mopicli -sv
Verbose status: 
Source # 1 
Source full active (blue LED) 
Source # 1 
Source good # 2 low / not present 
User configured

more @ ~ $ raspberryTEST dates
Fri 12 February, 2016, 00:12:43, CET
pi@raspberryTEST ~ $ date

ven 12 feb 2016, 00.27.04, CET
pi@raspberryTEST ~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- --
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
70: -- -- -- -- -- -- -- --

What else can I add except that it was really very courteous. If I can repay you in some way, do not hesitate to contact me.
Greetings and good luck to you :)

Flavio

[image]

hamishcunningham, 2016-02-12:
you're welcome! good luck with the project!
h

closed 2016-12-19 by hamishcunningham