hamishcunningham/pi-tronics #70: MoPi as scheduler/power controller?

XFer012, 2016-01-05:
Hello, I was looking for a "timed power source" for my RaspPi projects, that is, something which could power on and off the RaspPI automatically on schedule (example: ON every day from 8:00am to 7:00pm, OFF otherwise) without human intervention and without wasting battery power while off. I *thought* MoPi could do that (and acting as a power hub of sort, a nice plus) and ordered one. 

But, now it seems it can't act as a "power timer": I just read through all the User Guide and found no trace of such function. :( 

Looks like I physically have to press a button for 3 seconds to power on/power off the RPi. :( 
If that's the case, the MoPi is basically useless to me. :( 

Did I miss anything in the User Guide, or did I get it wrong from the start and MoPi can't act as a scheduler/timer? 

Thank you very much

Fernando

stoduk, 2016-01-05:
The MoPi can do exactly this, except that it doesn't know time (ie. isn't
an RTC).  So you can tell it to go to sleep or wake up in a number of
seconds from now, but you can't explicitly say eg. "turn on at 8am".  If
your application will be network connected then you have all you need (run
ntp to get real time, and working out how many seconds to delay power
on/off is trivial); if it isn't then you'll need an external RTC to get to
the same place.

I made a time lapse project using a RPi Model A with RTC (as it was not
network connected).

See code and hackaday project below, former probably more interesting to
you.

HTH,
Anthony

https://github.com/stoduk/tempus_fugit
https://hackaday.io/project/3288-long-term-time-lapse

XFer012, 2016-01-05:
Hello Anthony,

first of all, many thanks for your support!

I downloaded your sources, and found it to be quite complex (of course, being a complex project).
Moreover, I have zero Python knowledge. :(
If I understand at least a bit, it looks like you exploit some sort of MoPi API?
I didn't find anything in the user guide!
Where should I look to know about MoPi "programming"?

To me, simple functions like for example "mopi.turnOffAfter(seconds)" or "mopi.nextPowerUpAfter(seconds)" are all I need!  Then I could wrap up everything in Bash or C.
But I didn't found anything in the docs!

As for knowing the time, no problem, I already have an RTC (DS1337) hooked up via I2C.

Thanks a lot!

Fernando

stoduk, 2016-01-05:
[FYI, I am just a nosey user, not one of the MoPi team :) ]

My code uses the MoPi APIs - but they expose a CLI as well.  I don't have
an RPi running at the moment, but I think it is pretty obvious - something
like "mopicli -won {number of seconds}" or "-woff {seconds}", to set the
power off/on delay.

Found it: https://pi.gate.ac.uk/pages/mopi.html#section-4.3, see mention of
-won/-wsd (not -woff!).

Only caveat is that if you set a power off delay then that is the time from
when MoPi next shuts down the RPi - but you can just set both on/off timers
at once.

HTH,
Anthony

XFer012, 2016-01-06:
Many thanks Anthony :)
Somehow I missed that page, doh.

The CLI combined with cron and RTC is everything I need, excellent!!

To MoPi team: it would be great if you could somehow "promote" the MoPi API and CLI in the docs, to better explain those timer-like functions. :) 

Now I can't wait to have the MoPi delivered!! :D

Fernando

hamishcunningham, 2016-01-06:
thansk anthony, hello xfer012

To MoPi team: it would be great if you could somehow "promote" the MoPi API
> and CLI in the docs, to better explain those timer-like functions. :)
>
do you know the "site:" query on google? I find
https://pi.gate.ac.uk/pages/mopi.html#section-4.2. as second hit or so for
the query "site:pi.gate.ac.uk mopi api" :-)

good luck!
best
h

XFer012, 2016-01-06:
Is that an answer?
Seriously?

First of all, I did not know a "MoPi API" would exist, so what to search for?
I only came to know about a MoPi API thanks to Anthony.

You could have highlighted those interesting functionalities much better, and could have expanded that 4.2 half-page with examples, use cases, scenarios.
I have almost stumbled upon MoPi while looking at Raspberry timers: you don't promote MoPi as the powerful swiss army knife tool it is, because you only focus upon "power hub" functions. Which are useful, no doubt, but since there are no power timers out there (and yes, I purchased and tested WittyPi...), I would have screamed about it with all my voice.
There are a buckload of battery-operated RPis out there, and a good chuck may benefit of scheduled power cycles. Add an RTC to the board and you would have a killer gizmo.
 
That said, if you're satisfied with the current status of documentation and like to be rude with your customers, hey, be my guest... it's your product.

Fernando

hamishcunningham, 2016-01-06:
is that a question?

closed 2016-01-22 by hamishcunningham