hamishcunningham/pi-tronics #71: MoPi attached, but no longer communicating I2C

RZRZR, 2016-01-17:
Hello,

Have been happily using the MoPi all day. Have been ssh'd into it. Pi is still powering up ok, battery is still at a healthy 12.50V, but I can't communicate with the I2C bus.

```mopicli. I2C bus input/output error on read word. Check bus? Check connection?```

I have turned it off, removed from Pi and disconnected from battery. Put it all back to together again and getting the same error.

What's happening?

RZRZR, 2016-01-17:
```
pi@raspberrypi:~/balcony $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --     
```

hamishcunningham, 2016-01-17:
hi, I'm off sick but seeing as it's you :-)

That's a new one!

Have you tried a different Pi? Was there some physical trauma?!

Lubo says:

I2C bus is direct connection between the RasPi processor and MoPi's
microcontroller. Do not believe that some of these port pins have burned. The
load is small. Is it possible the OS driver configuration to change suddenly?
I2C driver to be not loaded? Speaking honestly do not have any good idea what
is happening ....

I2C bus is terminated by 2 pull-up resistors on the Pi PCB. If one of them is
missing, no communication. Some cold joint there? Or at our end, on one of the
port pins?

RZRZR, 2016-01-17:
Hello!

Thank you - sorry you're sick :(

No physical damage - and I did try with a different Pi.

It's good to know that damaging the chip is unlikely.

I don't know how I could have knocked the driver out of wack, but I will
prepare a new install of Raspbian this evening and see if that fixes it.

Will report back

Thanks.
R

stoduk, 2016-01-17:
Just to check - have you got the right I2C bus address?  ie. do you still
get no response if you use "-y 0" instead?

RZRZR, 2016-01-17:
Yup - still no response on '-y 0'

Setting up a new SD card now

RZRZR, 2016-01-17:
Hmmm... So. Haven't swapped the SD cards yet..

Uncommented the I2C sections of /boot/config.txt
Rebooted and I get something on i2cdetect.

Run my script, get error and then get nothing. Done this twice with the same result.

This is a gist of the db file that the data.py is calling from: https://gist.github.com/RZRZR/0539093463e6749be37a.js

```
pi@raspberrypi:~ $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- 0b -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --                         
pi@raspberrypi:~ $ cd balcony
pi@raspberrypi:~/balcony $ sudo python3 data.py
Traceback (most recent call last):
  File "data.py", line 2, in <module>
    import db
  File "/home/pi/balcony/db.py", line 10, in <module>
    mp = mopiapi.mopiapi()
  File "/home/pi/balcony/mopiapi.py", line 32, in __init__
    [self.maj, self.minr] = self.getFirmwareVersion()
  File "/home/pi/balcony/mopiapi.py", line 147, in getFirmwareVersion
    word = self.readWord(0b00001001) # 9
  File "/home/pi/balcony/mopiapi.py", line 175, in readWord
    return self.baseReadWord(register)
  File "/home/pi/balcony/mopiapi.py", line 167, in baseReadWord
    if e.errno == errno.EIO:
UnboundLocalError: local variable 'e' referenced before assignment
pi@raspberrypi:~/balcony $ sudo i2cdetect -y 1
     0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
00:          -- -- -- -- -- -- -- -- -- -- -- -- -- 
10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- 
70: -- -- -- -- -- -- -- --    ```

hamishcunningham, 2016-01-18:
that gist link doesn't work for me... what is the script doing?
h

RZRZR, 2016-01-18:
My bad - Gist here - https://gist.github.com/RZRZR/0539093463e6749be37a

Rachel Rayns

Creative Producer

Raspberry Pi Foundation
UK Charity No. 1129409
Email: rachel@raspberrypi.org

WWW: www.raspberrypi.org

guruthree, 2016-01-18:
Well this is weird. Before running your script, does mopicli function correctly, i.e., can you run `sudo mopicli -v1 -v2`? Multiple times?

RZRZR, 2016-01-18:
I will test this when I get home.

@p33m asks if we could have a schematic to look at? The one of the site is too low res to read whats what.

rachel@raspberrypi.org

RZRZR, 2016-01-18:
Thanks for the schematic - came home and found Pi off. On MoPi button press it flickered two amber and the RGB on blue for a split second then off and smell of magic smoke. Now the Pi (without MoPi) plugged into normal USB power gets very very hot and doesn't boot.

@p33m and I will have a look tomorrow, but I think he's dead, Jim.

hamishcunningham, 2016-01-19:
Hmm. In my considered opinion I would say that there's a fubar gizmo in there somewhere...

Shall I send you another MoPi? (Where did you get that one from, btw? And stackable or low profile?)

RZRZR, 2016-01-19:
Thanks - We are testing things today - I think @p33m may report back with
findings.
I got it from the good ship Pimoroni - Stackable.

Rachel Rayns

Creative Producer

Raspberry Pi Foundation
UK Charity No. 1129409
Email: rachel@raspberrypi.org

WWW: www.raspberrypi.org

RZRZR, 2016-01-23:
Ok, new Pi, new MoPi (thankyou Hamish):

```
pi@raspberrypi:~/balcony $ python3
Python 3.4.2 (default, Oct 19 2014, 13:31:11) 
[GCC 4.9.1] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import mopiapi
>>> mp = mopiapi.mopiapi()
>>> batt = mp.getVoltage(1)
>>> print(batt)
0
>>> batt2 = mp.getVoltage(2)
>>> batt2
12246
>>> 
pi@raspberrypi:~/balcony $ sudo python3 db.py
Traceback (most recent call last):
  File "db.py", line 10, in <module>
    mp = mopiapi.mopiapi()
  File "/home/pi/balcony/mopiapi.py", line 32, in __init__
    [self.maj, self.minr] = self.getFirmwareVersion()
  File "/home/pi/balcony/mopiapi.py", line 147, in getFirmwareVersion
    word = self.readWord(0b00001001) # 9
  File "/home/pi/balcony/mopiapi.py", line 175, in readWord
    return self.baseReadWord(register)
  File "/home/pi/balcony/mopiapi.py", line 167, in baseReadWord
    if e.errno == errno.EIO:
UnboundLocalError: local variable 'e' referenced before assignment
pi@raspberrypi:~/balcony $ sudo mopicli -v -v1 -v2
mopicli. I2C bus input/output error on read word. Check bus? Check connection?
pi@raspberrypi:~/balcony $ 
```

And it then needs rebooting. The code is knocking out something - https://gist.github.com/RZRZR/0539093463e6749be37a

Will start commenting bits off and see what it is.

RZRZR, 2016-01-23:
Strike that last comment - I was running something in the background.

New MoPi is working well now. Have fuses in place incase this solar controller plays up. Will start putting together a tutorial tomorrow.

R

P33M, 2016-01-23:
R.E the post-mortem on the first MoPi unit:

The failed components in the setup were the MoPi, the attached A+ and attached camera module. The camera module had significant damage around U1 (3v3 to 2v5 LDO) and the A+ had a direct short between SoC 5V input and ground. The likely cause of these faults (though not proven) is that MoPi somehow supplied a significantly higher voltage than 5V to the attached Pi.

On testing the MoPi in isolation, the onboard MCU is fried. Frigging out the enable pin that the MCU uses to gate the onboard buck converter resulted in stable 5V output. No other devices in the power chain were damaged, as far as I can tell.

I can't determine what caused the chain of dead components - usually the culprit is the biggest smoking hole out of all the other smoking holes, but MoPi seems to have suffered the least amount of damage.

RZRZR, 2016-01-24:
Another Pi blown. Will give to @P33M to look at tomorrow.

I'm getting MoPi to turn the Pi off at **.15 pass the hour and reboot at the following **.00. In those 15 minutes it takes a photo and uploads it to dropbox (scheduled for **.02). It also takes V reading from MoPi, puts it in a cvs file and syncs that to dropbox (scheduled for **.03). It happily did this from 22.00 yesterday to 11.03 today... but it didn't shut down at 11.15 or record any Volts data at 12.03. 

Is there an issue with frequent shutting on/off?
Controller is *claiming* to be giving MoPi a steady 12.1-12.4V, which MoPi confirmed to be true until it's last reading at 11.02... 

I don't know if the solar panel is getting enough sun at noon to actually be charging the battery (its really overcast here - it could well have been noon before there was enough sun for panel to be useful)... and that is causing the controller to send much higher Vs to MoPi.

I have a 5amp fuse between controller and MoPi that is still in tact  (receiving some 1amp, 2amp and 3amp fuses in the post tomorrow).

I working with a B at the moment, I have another couple I can use, but would like to try and identify what might be the problem before sacrificing them to the angry solar controller god.

RZRZR, 2016-01-24:
Tested on another Pi with bench power supply to the MoPi - MoPi still lives and mopiclic - v is giving me consistent volt readings, although ~0.5V less than bench supply's readout.

LuboBonchev, 2016-01-24:
>> Tested on another Pi with bench power supply to the MoPi - MoPi still lives and mopiclic - v is giving me consistent volt readings, although ~0.5V less than bench supply's readout.
The two power inputs are ored via Shottky diodes. Please see the schematic diagram. The voltage drop is 0.4 - 0.5V. That's why you see different readouts.

LuboBonchev, 2016-01-24:
Regarding the I2C issue: Is it possible to have voltage spikes on the inputs, higher than 28V as example? And if yes, how long such spike can be?

LuboBonchev, 2016-01-24:
And is it possible to have underspikes, lower than 6V?  If yes, how long such spike can be?

RZRZR, 2016-01-24:
I will let @p33m answer those second ones. I know the Max V from solar panel is 17V, but the controller I don't know. Jon has poked at its insides, he will be able to answer

guruthree, 2016-01-25:
Hmm. Dying in the middle of the day? When I was testing with 10 W 17 V solar panel and the MoPi/Pi, I found that on a fairly bright day that the 17 V panel would peak at a much higher output of around 19 V (under the modest load of a Pi) in direct sunlight. For me with the panel directly attached to the MoPi, that's was getting very close to the point where the MoPi starts to burn out around 21.5V. Maybe there are some similar high-voltage shenanigans with the controller?

P33M, 2016-01-25:
From the schematic, it appears that voltage sensing is taken from before the two mixing diodes via R4 and R5 (hence having two independent measurements).

It's highly unlikely that significant voltage variations are occurring as the battery is effectively directly connected to the load terminals (via low-voltage shutoff FET) inside the controller. A 20W panel isn't going to push 7.2Ah 12V battery voltage around particularly fast. The maximum charge voltage is 14.4V in the equalisation phase.

LuboBonchev, 2016-01-26:
"From the schematic, it appears that voltage sensing is taken from before the two mixing diodes via R4 and R5 (hence having two independent measurements)."
Correct. An d the firmware deduct the measured values taking into account the diodes. 

I do not have any bright ideas, sorry :-) This case is the first.I do not know similar one. Can't reproduce the case here because do not have the same equipment. Just like to say that all MoPis passed functional tests before the export. Also, Fred did a lot of MoPi testing with solar panels and no one unit has burned ....

hamishcunningham, 2016-02-02:
hope the move went well

I think the only credible answer is that the solar controller is stuffing an unfeasible voltage up the rig?

in which case you might invest a couple of pounds in a 12V UBEC and try again?

hth, h

closed 2016-12-19 by hamishcunningham