hamishcunningham/pi-tronics #72: MoPi power sources flash 10 times on power up

swojtowicz, 2016-01-24:
Hi, my MoPi was working fine.  Suddenly when I try to turn it on, the 2 power source LED's flash about 10 times then stop.  Nothing else happens and it won't turn on.  I checked the voltage out from my 2 power supplies (9v's) and they both are about about 7.8 volts.  Is there any meaning behind the flashing LED's?

Thanks!

hamishcunningham, 2016-01-24:
perhaps the 9v supplies are dipping under load?
it sounds like there is insufficient voltage to persuade mopi to start -- that could also be a config error; did you config simbamond for your supply characteristics?
hth
h

swojtowicz, 2016-01-24:
Aahhh I'm dumb.  It was the batteries.  As soon as you said they might be dipping "under load", I went and picked up more 9v's (had no spares to try when I first posted my issue).  I should have tried that first.  Thank you!

hamishcunningham, 2016-01-25:
Cool, have a good one!

closed 2016-01-25 by hamishcunningham