hamishcunningham/pi-tronics #74: Rainbow Square

marty-sullivan, 2016-02-09:
Hi, I have [this battery](http://www.amazon.com/gp/product/B011BGPUSI?psc=1&redirect=true&ref_=oh_aui_detailpage_o02_s00) and it seems to power the MoPi OK at whatever voltage I set the DC output. The MoPi status also reports the correct input voltage from the battery. However, I consistently get the dreaded "Rainbow Square" from my Pi 2 whenever it is running, signifying under voltage from the MoPi. I don't have any usb devices plugged in, just HDMI display. Is there anything I should check in the config or otherwise?

hamishcunningham, 2016-02-10:
hi Marty,

The battery *should* be ok; I'd recommend 12V if you're not using a second
supply.

It is possible that your Pi is unusually fussy about the voltage -- MoPi's
output is a fraction under 5V in order to
allow directly connected supplies to overpower it. This doesn't create a
problem in our experience.

Does the rig actually work normally?

HTH, Best

Hamish

marty-sullivan, 2016-02-10:
Thanks for the info. Yes, the Pi seems to run normally. I can run things fine with the ARM at 400%, no voltage warning. It consistently warns me any time I do something that suddenly draws more power, but when running at full CPU for a length of time I get no warning.

closed 2016-12-19 by hamishcunningham