hamishcunningham/pi-tronics #75: mopi hangs after auto-power-off

milode, 2016-02-12:
Hi, I have problems with mopi after auto-power-off from mopi when battery is discharged. Tried 2 different mopis. both hang reproducable after first auto-power-off:

I use 2 voltage sources:
source1= DC: 18.4V (from ac/dc converter, ripple very low)
source2= 12V/1.2Ah Pb-accu: charging Voltage=14.4V  Standby Voltage=13.3V 

1) New HW installation (powersources disconnected an connected again)
   after switching on -> source1 amber-LED and source2 amber-LED is continuously on
   switching between source1 and source2 works
   status s_bat_max, s_bat_full, s_bat_low, s_bat_critical from statusword is correct
   shutdown and power switch off by mopi works correct whenn battery reaches s_bat_critical
   Mopi-status-LED (blue, green, red) works

2) after first auto-switch-off by mopi (battery discharged)
   I connect DC-power (18.4V) again, -> battery is loading
   I switch on raspberry with mopi-button (3 sec)
   source1 amber-LEDis continuously on, source2 amber-LED is blinking
   after 5 hours battery is full but source2 amber-LED is blinking
   Switching off and on the raspberry with mopi-button
   -> source1 amber-LEDis continuously on, source2 amber-LED is blinking
   switching between source1 and source2 works
   mymopi.getVoltage(2)-values are correct
   rc2[]-values are correct
   status s_bat_max, s_bat_full, s_bat_low, s_bat_critical from statusword is WRONG shows always s_bat_full
   shutdown and switch off by Mopi DOES NOT WORK if battery is critical (battery will be completely discharged until ~6V)
   
   charging battery again -> source1 amber-LEDis continuously on, source2 amber-LED is blinking
   I unplug source1 (DC) -> source1 amber-LEDis and source2 amber-LED are off
   every ~x minutes source2 amber-LED blinks some seconds and gets off again
   I type sudo shutdown -h now -> raspberry goes down
   I try to switch off power with mopi-button (3 sec) -> DO NOT WORK (red mopi status-LED is blinking but no power switch off)

   the only thing that brings back mopi to normal operation: Unplug and replug DC and battry
   after switching on with mopi-button (3 sec) -> everything works correct again
   but after next auto-shutdown by mopi after battery discharge -> same problem again

   tried two different mopis serial 862 and 876: both same behaviour

   looking for transient overvoltages (on source1 and source2) with oscilloscope but there was nothing to found also during DC-on/off with full or discharged battery

    Can you help?
------------------------------
HW-/SW-Versions I use:
------------------------------

MOPIs:
Mopi1: Serial number: 862, Firmware version: 3.10
Mopi2: Serial number: 876, Firmware version: 3.10
Mopi3: Serial number: 871, Firmware version: 3.10 (not tested)

RASPBERRY
Raspberry Version: ->> Model B Rev 2
pi@RasPis ~ $ cat /proc/cpuinfo
processor       : 0
model name      : ARMv6-compatible processor rev 7 (v6l)
BogoMIPS        : 2.00
Features        : half thumb fastmult vfp edsp java tls
CPU implementer : 0x41
CPU architecture: 7
CPU variant     : 0x0
CPU part        : 0xb76
CPU revision    : 7
Hardware        : BCM2708
Revision        : 000e
Serial          : 00000000f6915377

SIMBANOND:
simbamond 4.1 

SIMBANOND(config):
...
#############################################################################
# local config - DON'T EDIT THIS LINE!
# wc1: -wc1 3 18500 18000 6800 6200
# wc2: -wc2 1 12400 11600 10900 10400
# end of local config - DON'T EDIT THIS LINE EITHER!
...

MOPICLI:
# Version of mopicli
CLIVERSION=0.5
# Version of API used
APIVERSION=0.3
# Package version
VERSION="4.1"

LuboBonchev, 2016-02-17:
Hi milode,
Thank you for willingness to use our product. Thank you for complete report of the issue you have. MoPi is designed to work mainly with batteries and strictly is monitoring the discharged battery to be replaced, meaning TO BE DISCONNECTED AND NEW ONE TO BE CONNECTED. Charging of the battery at place is not expected. I think this is the root of your issue. 
If you have further questions, please do not hesitate to ask us again. 
Regards,
Lubo Bonchev
SelCom Ltd.

milode, 2016-02-17:
Hi Lubo,
thank for reply.
first I was realy confused, thougt something is damages. But after the
experience that it works again after discennection I supposed that this
is the way it works.
My solution now is to insert a relay which disconnects the low battery
if raspberry is down.
I suppose this should work.

Best Regard
Michael

hamishcunningham, 2016-02-17:
yes, sounds like a good workaround :-)
h

Paul-B, 2016-03-06:
Hi all,

If I read this:
TO BE DISCONNECTED AND NEW ONE TO BE CONNECTED. Charging of the battery at place is not expected. 

Then am I correct that the expectation comes from the software? Because if that is true then I could maybe tell the software to not behave like that. If I install a RPi in my car with MoPi it is fairly safe to assume that the car systems will charge the car battery while in use, also you could still monitor the voltage of the car battery.

Regards,

Paul

hamishcunningham, 2016-03-06:
that's in firmware :-(
h

Paul-B, 2016-03-06:
Fair enough.
I should probably work with relays then, unless this is a way out:
Monitor the car's battery voltage from the Pi and don't let the voltage drop to a critical level but enough to run for about an hour or so. Then shutdown the Pi from a script.
Could this work without a relay or must I still disconnect the car's battery before the next start?

LuboBonchev, 2016-03-06:
Should work. Use the yellow LED as criteria. Until it lights stable, should work. If begin to flash, the firmware is looking for battery change.
Lubo

Paul-B, 2016-03-06:
Thanks for the input!
Perhaps I could monitor the battery voltage through mobicli. Allow the battery to go down to a certain voltage and then call for a shutdown.

LuboBonchev, 2016-03-06:
Yes, you can.
   mopicli -v1 or -v2
if I still remember the commands :-)

LuboBonchev, 2016-03-06:
and when trim the voltage level in your program, keep the eye on yellow LED :-)

Paul-B, 2016-03-06:
Well this page
https://pi.gate.ac.uk/pages/mopi.html#cli
Suggest you got the command right. I can indeed read a battery with it. 
I'm still a python n00b but it looks a lot like php so I should be able to pull it off.
In the end I won't be monitoring the led's because I want the hide the Pi behind the ceiling panel of my Smart and install an auto-booting dashcam.
Already have everything else working. I need the delay to give the Pi time to connect to my home wifi and transfer the files to my NAS via bittorrent sync. Works like a charm on my experimenting table. Not draining the car battery and auto-boot is my missing link.

LuboBonchev, 2016-03-06:
Have a good luck then. If more help needed do not hesitate to ask me.

Paul-B, 2016-03-06:
Thank you.
Glad to hear that what I want seems possible.

hamishcunningham, 2016-03-07:
hi,

I think I just discovered a workaround -- if you push config to the MoPi it
appears to reset the LED status :-)

h

Paul-B, 2016-05-13:
So I finally had time to do some testing and found that indeed even if you shutdown the pi from a script you cannot just start it again, the battery must first be disconnected.
I read your previous remark Hamish and I wonder what you mean by that. Correct me if I'm wrong but if the Pi was shut down how do you do that? Or maybe I should better ask, at what point (and how) would you push the config to the MoPi?

closed 2016-12-19 by hamishcunningham