hamishcunningham/pi-tronics #76: What is the current draw when the Pi is not running?

tamaskoloti, 2016-03-30:
Hi,

I'm working on a project that requires the PI to turn on a couple of times a day for a few minutes, leaving the PI off for most of the time. It will use a 1x 8xAA battery pack. MoPi seems to be the perfect solution for me because it has all the features I need.
Ideally I would like this setup to work for at least a month for a single charge. I'm wondering if there is any information/measurement about the current draw when the PI is off? This would be crucial for my project to estimate/guess how long it can run without changing batteries. I read the user guide and went through the issues here but I couldn't find any information about this.

Can you please help me with this?

Thanks,
Tamas

stoduk, 2016-03-31:
Hi,

[Hamish and team might be able to give you accurate standby power
consumption figures, I can just give some arm waving answers that might
help in the mean time.  I don't recall seeing the numbers you want
published anywhere]

I had a camera project which would turn on every half an hour during
daylight hours, just to take a picture (so 19 power cycles, from 0730 to
1630).  Using 8xAA NiMH I got less than a week on each set of batteries (5
days, depending on temperature) - but from testing the vast majority of the
battery usage was from the times when RPi was running.

If you really are only running twice a day for a few minutes then that is
10x less often than I did, so you might expect 10x longer battery life.  So
 a month may be possible.  However, I'd recommend:

1) testing! Power usage is hard to predict, in particular the effect of
self-drain of your batteries, how temperature fluctuations will impact
usable capacity, etc.

2) Pick your RPi - those running USB networking, etc. are far more power
hungry.  [I was using a model A, the lowest power at the time]

3) minimise boot time.  I never went down the path of rolling my own RPi
image, but having the leanest software install should improve your chances
by improving startup/shutdown delays.

HTH,
Anthony

tamaskoloti, 2016-04-03:
Hi @stoduk,

Thank you for you reply. I know that testing is the best way to figure out. I was just curious if MoPi's standby mode was designed for this kind of scenario. I checked the photos of the circuit board (the picture of the circuit diagram is not very helpful due to a low resolution) and it seems that it uses an ultra low power version of a 8051 compatible cpu. I guess if the standby power regulator is efficient too then this won't be an issue. I'm hoping that @hamishcunningham will confirm this.

From the power on perspective I use a Raspberry PI zero, so hopefully it will save even some power compared to model A. This is an in-door project, so the temperature won't affect it that much.
Thank you for the boot time optimisation suggestion! I will definitely look into that.

Thanks,
Tamas

hamishcunningham, 2016-04-04:
I haven't got anything to add to Anthony's reply -- good luck!
Cheers
Hamish

closed 2016-04-04 by hamishcunningham