hamishcunningham/pi-tronics #77: dead mopi

gituseraccount, 2016-04-01:
My mopi is not working correctly. I bought it from buyapi They redirected me to contact you. I'm not very good . Programming I don't see what's wrong. I just need the mobile power.

hamishcunningham, 2016-04-01:
pls detail what you did, and what "not working" means. thanks!

gituseraccount, 2016-04-01:
Simply does not work ie: no power no voltage , current and resistance with volt meter

From: Hamish Cunningham
Sent: Friday, April 1, 2016 5:15 PM
To: hamishcunningham/pi-tronics
Reply To: hamishcunningham/pi-tronics
Cc: gituseraccount
Subject: Re: [hamishcunningham/pi-tronics] dead mopi (#77)

pls detail what you did, and what "not working" means. thanks!

-

hamishcunningham, 2016-04-01:
how did you connect it? how did you configure it? what batteries did you
connect? did any lights come on? did they flash?
tnx
h

gituseraccount, 2016-04-01:
Through the power sockets. Used this guide http://projects-raspberry.com/mopi-is-mobile-hot-swap-and-247-power-for-the-raspberry-pi/ A Put it with a 3.72000mah - 5volt battery with 3000ma. Also tried other batteries so it's not that for sure and it's not the raspberry pi 3.

From: Hamish Cunningham
Sent: Friday, April 1, 2016 6:24 PM
To: hamishcunningham/pi-tronics
Reply To: hamishcunningham/pi-tronics
Cc: gituseraccount
Subject: Re: [hamishcunningham/pi-tronics] dead mopi (#77)

how did you connect it? how did you configure it? what batteries did you
connect? did any lights come on? did they flash?
tnx
h

--

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK
+44 7920 765 455
hamish@gate.ac.uk https://twitter.com/@HCunningham
https://hamish.gate.ac.uk *http://aquaponicslab.org/
<http://aquaponicslab.org/>*
https://gate.ac.uk https://pi.gate.ac.uk
http://www.dcs.shef.ac.uk/cgi-bin/makeperson?H.Cunningham

-

hamishcunningham, 2016-04-02:
MoPi requires at least 6.2V to run. Try connecting a 12V battery for
example.

See e.g. https://pi.gate.ac.uk/pages/mopi.html#section-2.3.

HTH, best
Hamish

gituseraccount, 2016-04-02:
still doesn't work
From: Hamish Cunningham
Sent: Saturday, April 2, 2016 7:14 AM
To: hamishcunningham/pi-tronics
Reply To: hamishcunningham/pi-tronics
Cc: gituseraccount
Subject: Re: [hamishcunningham/pi-tronics] dead mopi (#77)

MoPi requires at least 6.2V to run. Try connecting a 12V battery for
example.

See e.g. https://pi.gate.ac.uk/pages/mopi.html#section-2.3.

HTH, best
Hamish

--

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK
+44 7920 765 455
hamish@gate.ac.uk https://twitter.com/@HCunningham
https://hamish.gate.ac.uk *http://aquaponicslab.org/
<http://aquaponicslab.org/>*
https://gate.ac.uk https://pi.gate.ac.uk
http://www.dcs.shef.ac.uk/cgi-bin/makeperson?H.Cunningham

-

gituseraccount, 2016-04-20:
Ya if you can send replacement
I send it back
106 niagara  st unit 2
St catharines
Ontario
Canada
L2r4l4

From: mike o
Sent: Saturday, April 2, 2016 4:11 PM
To: Hamish Cunningham; hamishcunningham/pi-tronics
Subject: Re: [hamishcunningham/pi-tronics] dead mopi (#77)

 still doesn't work
From: Hamish Cunningham
Sent: Saturday, April 2, 2016 7:14 AM
To: hamishcunningham/pi-tronics
Reply To: hamishcunningham/pi-tronics
Cc: gituseraccount
Subject: Re: [hamishcunningham/pi-tronics] dead mopi (#77)

MoPi requires at least 6.2V to run. Try connecting a 12V battery for
example.

See e.g. https://pi.gate.ac.uk/pages/mopi.html#section-2.3.

HTH, best
Hamish

--

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK
+44 7920 765 455
hamish@gate.ac.uk https://twitter.com/@HCunningham
https://hamish.gate.ac.uk *http://aquaponicslab.org/
<http://aquaponicslab.org/>*
https://gate.ac.uk https://pi.gate.ac.uk
http://www.dcs.shef.ac.uk/cgi-bin/makeperson?H.Cunningham

-

closed 2016-12-19 by hamishcunningham