hamishcunningham/pi-tronics #78: 4.66V on the GPIO

BimberX, 2016-04-02:
MOPI is a tool I have been looking for to use in my project.
I have it connected in RPi v2, with additional cables soldered to pin 2 and 3 of GPIO to power the external HUB. Unfortunatelly I have noticed few hangups and power shortage leading to hangs of the Pi. So I decided to measure the voltage on the GPIO pins. It came out 4,66V which is far too less for RPi to work properly and give power to USB ports (not the mention external HUB).

Is there a way to make Voltage go up to 5.1V (just as it is to go down to 3,3V)?

Would be gratefull for advice.

hamishcunningham, 2016-04-04:
If you're trying to run a lot of peripherals then there may not be enough power (max current is 2A).

Also note that MoPi intentionally sets its voltage at a little less than the 5V on the Pi's micro USB connector, so that the latter will overpower it -- this is so that you still have the option to connect a supply via that connector. This works fine :-)

So: my best guess is that you're trying to pull more current than we can supply via your external hub :-(

If you need to slave another PSU, you could use a circuit like this: https://www.circuitlab.com/circuit/wk2g43/mopi-high-current/

HTH, Hamish

closed 2016-04-04 by hamishcunningham

BimberX, 2016-04-04:
Thank you for quick answer.
As for now I have 4,70V on Mopi output with current of 0.14A. To power any of my peripherials (3G modem and WiFi card) I need 5V stable the problem is the lack of Volts not the lack of Amps :)
I had the same problem when I connected a charger that was giving 4.8V directly to Pi, so I know this is voltage problem.

The diagram looks very promissing. I'll try to build it and let you know.

hamishcunningham, 2016-04-04:
Hard to know what is wrong then, as there are several thousand MoPi out there that are working well... I mention current as if you try to draw too much then the voltage will fall. Sorry I can't be of more help :-(

BimberX, 2016-04-04:
No problem. Its all about experimenting and fun :)
As for the diagram, I can recognize the types of parts but not what their specific parameters should be. Is there any list of specific parts shown on the diagram?