hamishcunningham/pi-tronics #83: Using MoPi as UPS

occuPi-d, 2016-05-21:
> MoPi can be used as a UPS (Uninterruptible Power Supply) in two ways. If the Pi is powered over its 5V micro-USB socket and that power drops while MoPi is active, the supply fails over transparently, allowing the Pi to continue work uninterruped

With the following configuration:
- Single 9V battery to MoPi
- Standard 2A power supply to Raspberry Pi

The 9V battery drains out very quickly even though the Raspberry Pi is mains powered. Is this correct behaviour? I've tried a couple of different Raspberry Pi models with the same result.

Many thanks

hamishcunningham, 2016-05-27:
If you're using a PP3 note that these aren't really up to the job of supplying a Pi; I've had them work for a short while in the past but gave up in the end. This will be more true with Pi 3 and/or running peripherals. But no, the mains supply should overpower the battery when it is connected. Do you have a multimeter? Check the voltage from the PSU, and the current on both lines?
HTH
H

occuPi-d, 2016-05-31:
Many thanks for the reply. I've tried it with 9V PP3 and also 6 x AA.

The problem isn't whether the batteries are able to power the Pi, it is that the batteries discharge even when the Pi is mains powered.

I've been cross-referencing the status page of MoPi with the output of various PSUs. When the PSU output is slightly less than 5V (common albeit not ideal) the rate of battery discharge is much quicker. When 5.1V is mains supplied, the rate of battery discharge is minimal.
_I should add: all these PSU's will power the Pi without any apparent ill effects, even if outputting slightly below 5V._

I am wondering what is the threshold for the MoPi to step in, and can this be modified? i.e. Is it the mains supply falling below a certain voltage, or when the mains supply voltage starts to decrease etc.

Many thanks once again.

hamishcunningham, 2016-06-01:
I'm afraid we're at the mercies of the vaguaries of 5V PSUs :-(

changing the threshold requires reflashing the firmware, which is not
currently an end-user option :-(
(hopefully it will be in the nextx version)

can you use a 5.1V PSU in your application?

cheers

h

-- 

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK

BBC Radio Sheffield say I'm a *fish poo boffin*
(http://www.bbc.co.uk/programmes/p03v9nyv 2 hours 37 mins in;
also on vid: http://bbc.in/20x2C5J & scroll down a little)

Phone: +44 7740 149 842
hamish@gate.ac.uk                 https://twitter.com/@HCunningham
https://hamish.gate.ac.uk           *https://wegrow.social
<https://wegrow.social/>*
https://gate.ac.uk                  https://pi.gate.ac.uk
http://www.dcs.shef.ac.uk/cgi-bin/makeperson?H.Cunningham

occuPi-d, 2016-06-04:
Thanks for the reply once again. Ultimately, I wanted to use the MoPi as a way to cleanly shutdown the Pi once a standard USB power bank (connected directly to the Pi) is depleted. Unfortunately, the output of these powerbanks also dip below 5V too.
Oh well - looking forward to the next version with flashable firmware! 
:)

closed 2016-12-19 by hamishcunningham