hamishcunningham/pi-tronics #88: Mopi and USB battery pack

apr-1985, 2016-08-24:
Hi I have just bought a Mopi, and successfully powered with a 12v mains connection and a 9v battery backup.
However I would like to power from 2X 20000mAh power banks (as used for mobile phone charging). Each of which have USB outs.
I have tried stripping down a USB lead to its red and black wire and attaching that to the mopi and the USB end to a mains->USB adapter plug, however the mopi gets 0v from it.
(I tested that the cable and the main->USB plug could power the pi before chopping off the micro USB end so I know that works).

Any help most appreciated.
Adam

hamishcunningham, 2016-08-24:
Min volts in is 6.2!

apr-1985, 2016-08-24:
D'oh there goes that idea then. Back to the drawing board.

closed 2016-12-19 by hamishcunningham