hamishcunningham/pi-tronics #91: topi configuration - Error Message (mopicli. Expected at least MoPi firmware version 3.05, got 0.00 instead.)

RockSoft, 2016-10-11:
I have just purchased a MOPI for my raspberry pi zero and have installed
all the software components, using the online instructions.  I start the MOPI with out any problems 'Green LED is on' and both yellow LED's are on and stable. I have hooked up 2 9v rechargeable batteries.  

I am able to ssh into the raspberry pi zero without any problems.  I am then trying to run the topi configuration tool as per 2.3.2. Using the Configuration Tool on the Pi on the online user guide.

When I type sudo mopi in the terminal window I get the following message:

sudo mopi

/usr/sbin/mopi: /usr/sbin/mopicli not working; installation problem?
(mopicli. Expected at least MoPi firmware version 3.05, got 0.00 instead.)
Usage: mopi [-h(elp)] [-d(ebug)] [-v(ersion)] [-l(og entries)]

Any assistance you can give would be gratefully received.

hamishcunningham, 2016-10-11:
Hi,  MoPi uses the Pi's i2c bus, which, on the zero has to be soldered in... Did you do that? 
Cheers Hamish

RockSoft, 2016-10-11:
I have soldered 40 pins to the pi zero board and the topi is stacked on to those pins.  I also have external power taken through the GPIO pins from a second raspberry pi zero.  I have attached a picture of the setup.  I've just followed the online instructions that you have on your webpage.  so haven't done anything different.  There raspberry pi has been used for other things, but the I2C is enabled through the raspberry pi configuration.  I've attached the raps config for your information
[image]

[image]

RockSoft, 2016-10-11:
just ran: sudo mopi -d and it went into the configuration.  running the command without debug on -d I get the same error as before

hamishcunningham, 2016-10-11:
For some reason the i2c communication is not working, at least partially

What happens if you try and save config to the device?

RockSoft, 2016-10-11:
It says mopi has been configured, but when I go back into the config using the -d option 
[image]
the settings do not seem to have been saved.  When I go into local config the following is shown (see picture) My batteries have also died now, so won't be able to do any more on it until they have recharged :(  Are they supposed to drain while there is an external power supply?

hamishcunningham, 2016-10-15:
Do you have anything else that uses i2c that you could use to check the GPIO header?

RockSoft, 2016-10-19:
hi Hamish - I'm afraid I do not have any other hats I can attach to the raspberry pi.  However, I have tried attaching it to another pi zero, in order to see whether it might be a local GPIO issue but got exactly the same error message.  Could the MOPI itself be faulty or is there another way of testing the GPIO pins without a hat?

RockSoft, 2016-11-06:
Any update on this issue?  The following is a log from today

Nov  6 16:51:16 rockpi-0 logger: /usr/sbin/simbamon: /usr/sbin/mopicli not working; MoPi not attached or powered? (mopicli. Expected at least MoPi firmware version 3.05, got 0.00 instead.)
Nov  6 16:51:16 rockpi-0 simbamond[381]: Started simbamon.
Nov  6 16:51:16 rockpi-0 logger: simbamon: first run after boot, sleeping for 2...
Nov  6 16:51:16 rockpi-0 systemd[1]: Started LSB: simbamond: a simple battery monitor daemon.
Nov  6 16:51:19 rockpi-0 logger: simbamon: ...boot delay done
Nov  6 16:51:19 rockpi-0 logger: simbamon: writing local config /usr/sbin/mopicli -wc 1 8400 7500 6600 6000
Nov  6 16:51:20 rockpi-0 logger: simbamon: error writing config: mopicli. Expected at least MoPi firmware version 3.05, got 0.00 instead.
Nov  6 16:51:22 rockpi-0 logger: simbamon: MOPI_STATUS is mopicli. Expected at least MoPi firmware version 3.05, got 0.00 instead.
Nov  6 16:51:22 rockpi-0 logger: simbamon: invalid MOPI_STATUS (mopicli. Expected at least MoPi firmware version 3.05, got 0.00 instead.): will pause/retry...
Nov  6 16:53:42 rockpi-0 logger: simbamon: invalid MOPI_STATUS (mopicli. Connection timed out): will pause/retry...

hamishcunningham, 2016-11-07:
My apologies, but I don't have a zero to hand and can't try to replicate here :-(

I can offer you a refund?

RockSoft, 2016-11-07:
Ok.  I'll take a refund.  it's a shame its got great potential.

hamishcunningham, 2016-11-07:
fair enough

btw I'm not implying that I think the MoPi is definitely at fault here,
just that I don't have time to try and duplicate the problem :-(

please email me the sales invoice and your account details

thanks, and sorry I couldn't help!

h

-- 

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK

*Phone: 07740 149 842 *
hamish@gate.ac.uk            https://twitter.com/@HCunningham
https://hamish.gate.ac.uk    *https://wegrow.social
<https://wegrow.social/>*
https://gate.ac.uk           https://pi.gate.ac.uk
http://www.dcs.shef.ac.uk/cgi-bin/makeperson?H.Cunningham
https://www.sheffield.ac.uk/news/nr/aquaponics-fish-food-farm-1.581536
https://tinyurl.com/savetheworldwithfishpoo

WeGrow.social:
https://twitter.com/WeGrowSocial
https://www.facebook.com/WeGrowSocial
https://blog.wegrow.social
https://www.instagram.com/wegrowsocial
https://wegrow.social
https://www.youtube.com/channel/UCG3ZPbhbJ1yoa0fxMnaOcdg/videos

On BBC Radio Sheffield:

   - with John Grant: http://bbc.in/2ceFE57 (8 mins in)
   - https://hamish.gate.ac.uk/bbc-shef-aquaponics-may-2016.wav
   - https://www.facebook.com/bbcsheffield/videos/1214299238582354/

RockSoft, 2016-11-07:
Hi Hamish,

Not a problem and I fully understand.  

Sales invoice is attached, this includes the price for the pi zero as well.  Obviously, just the MOPI cost+VAT would be all you’d need to credit back.   Do you want me to send the MOPI back to you?

Account details are:

Rock Software Solutions Limited
Sort Code: 16-57-10
Account No: 55114477

If you could issue me with a refund notice then I can ensure that this is added to the quarterly accounts.

Many thanks

Peter Leigh – Managing Director

 +44 (0) 7872 347 944
Skype: peter.leigh.rockss
 peter.leigh@RockSoftwareSolutions.com <mailto:peter.leigh@RockSoftwareSolutions.com>
www.rocksoftwaresolutions.com <http://www.rocksoftwaresolutions.com/>

Rock Software Solutions Limited a company registered in England and Wales having company number 6881847
Registered Office: 71 Hibiscus Crescent,  Andover, Hampshire, SP10 3WF

closed 2016-11-14 by hamishcunningham