hamishcunningham/pi-tronics #94: Pi 3 undervoltage when using MoPi and 12v Pb Battery

frstep, 2016-11-28:
I'm getting under voltage (and therefore throttling) on both my Pi 3's when using MoPi. Various configurations tested, but always using with a I2C to 1-Wire host adapter. Without the MoPi, but using a USB power supply, the under voltage warning goes away and the CPU runs at 1.2GHz.

Is there an issue with using the MoPi with GPIO boards?

Regards,

F.

frstep, 2016-11-28:
Update: Under-voltage reported even without the 1-wire board. Both Pi 3:
        $ vcgencmd get_throttled
        throttled=0x50005 

Is this normal?

MelOcten, 2016-12-11:
I am experiencing the same problem on a Raspberry Pi 3 B V1.2 (newest software) with MoPi FW 3.10. Constant undervoltage warning when using MoPi with a 8 Volt PSU at source 1 configured as non-battery. The PSU can supply more than enough power for the rpi.

dmdelorme, 2016-12-11:
yup me too

frstep, 2016-12-11:
I think The MoPi pre-dates the Pi3, and clearly not able to supply the current required. The sellers should be aware of this :-(

hamishcunningham, 2016-12-13:
I've used a Pi 3 without getting a warning, but yes, there can be throttling, which seems to depend on the peripherals attached and the load it is running...

frstep, 2016-12-13:
I'm seeing throttling to 600 Mhz without any USB devices plugged in, only ethernet. And no significant s/w load. I have tested it on two Pi3. Software is Rasparian Jessie, relatively freshly installed. All this results in a Pi3 is slower than my Pi2. If you can suggest what I should check, I'm happy to do more tests, but I'm not the only person reporting this problem, which is curable simple by switching to the official power supply.

F.

> 
> I've used a Pi 3 without getting a warning, but yes, there can be throttling, which seems to depend on the peripherals attached and the load it is running...

--------------------------------
Fraser Stephens       
+44 (0)7966 412891  
+44 (0)1873 832499      
Skype: frstep

nimble99, 2016-12-19:
Ah, I have now stumbled across this thread too...

I purchased the Mopi to run Pi3 devices, and immediately discovered my USB peripherals would drop off the USB bus. I took the mopi to my engineer friend, and he's highlighted the fact that the Mopi supplies only 4.7V to a Pi3 at idle, and this drops to 4.3V when the total load (pi+peripherals) is 1.5A - so it appears the Mopi cannot supply the 2A stated on the website. 
How could this be related to Raspberry Pi 3 at all?  1.5A is 1.5A whatever the pi version is...

I am running a Logitech C920, and Huawei E3372 4G modem (powered directly from the MoPi - not the USB bus). Everything is fine for about 30 seconds, then the voltage sags way too low, and the C920 drops off.

Should this really be happening, if the MoPi can supply 2A?
The scopes showed no current spikes, and I had a 4700uF cap across the peripheral rails to smooth out any sudden draw coming from the 4G modem.

Any suggestions?

Thanks guys,

frstep, 2016-12-19:
Under voltage certainly causes problems with USB devices, which is why many people use a hub. In my case, my USB devices were happy with the MoPi, but the Pi3 throttles down to 600MHz, with or without the USB devices. I wonder how many people running MoPi on Pi3 have never noticed that they are getting less than the performance of a Pi2? I guess for many this is acceptable, since if you are powering off batteries, you are likely not to be after high performance and happy to trade performance with battery life.

dmdelorme, 2016-12-19:
I am Just a a little annoyed I guess the MoPi will go into the junk pile.. I hope the next version solves this. It would also be nice if it is mentioned on the site that this is not fully operational.

hamishcunningham, 2016-12-19:
Hi Folks,

You're right: we've decided to withdraw support for the Pi 3 on MoPi :(
Details here: https://pi.gate.ac.uk/posts/2016/12/19/good-news-and-bad/

The new (2.5A capable) version will be here soon. In the meantime,
I'm happy to give you a full refund -- just give me your bank details and a
copy of the purchase invoice. You can find my contact details here: 
https://hamish.gate.ac.uk/pages/contact.html

Sorry that we didn't get to this point earlier!

All the best, Hamish

closed 2016-12-19 by hamishcunningham