hamishcunningham/pi-tronics #95: Voltage displaying on raspberry pi 3

cpp5126, 2016-12-02:
currently, I am working on a project that uses Mopi to communicate and charge the Raspberry pi 3. I have trouble displaying the voltage on raspberry pi without using its buit-in api. Is there any other way we can display voltage on raspbery pi by ruuning a single python script to access the voltage data.

hamishcunningham, 2016-12-19:
Hi, the command-line API should supply you with voltage data?

Note: we aren't supporting MoPi wth the Pi 3 now :-( 
See https://pi.gate.ac.uk/posts/2016/12/19/good-news-and-bad/ for details.

Best, Hamish

closed 2016-12-19 by hamishcunningham

ronisegoly, 2016-12-24:
I just saw that Mopi does not support RPI3 
I have also RPI2 I will test with
When will you support RPPI3