hamishcunningham/pi-tronics #97: Where is the config file? (Raspberry)

ronisegoly, 2016-12-23:
I modified the config using mopicli but after reboot it returns to default

hamishcunningham, 2016-12-23:
Hi, what platform and versions are you using, and how do you verify that the change isn't persisted?

This is the script that stores and retrieves config data on the Pi (which is stored separately on the MoPi microcontroller): https://github.com/hamishcunningham/pi-tronics/blob/master/simbamon/mopi

It stores the Pi version of the data relative to the installation directory of the script -- so it depends on how you installed it...

HTH, H

closed 2016-12-23 by hamishcunningham

ronisegoly, 2016-12-24:
I ran 
mopicli -wc2 1 12000 10000 8000 6500
Then
mopicli -e
Made sure the above is the result for source 2
I rebooted
I ran mopicli -e again
The output is 
Source #2 conf: 
  Source type: Rechargeable batteries
  Maximum voltage: 11200 mV
  Good voltage: 10000 mV
  Low voltage: 8800 mV
  Critical voltage: 8000 mV

hamishcunningham, 2016-12-24:
Can you try it with the mopi command instead? Does the same thing but
should give you an error message if it fails to talk to the board

ronisegoly, 2016-12-24:
you mean the mopi interface (like raspi-config)?
I do not see how I can change the voltage level there
BTW I was on RPI 3, does it matter?