hamishcunningham/pi-tronics #98: Using solar panel as power source

ronisegoly, 2016-12-23:
I need to understand the usage of solar panel, if I combine battery and solar panel will it use solar power when available to save battery?

hamishcunningham, 2016-12-23:
Solar panel outputs are infinitely variable :-)

Assuming that you have one with an output voltage regulator integrated that supplies >~7V and less than ~20V then MoPi will choose the supply which has a higher voltage at any given time. So e.g. if you're running from a 12V battery, whenever the solar panel output is greater than 12V then MoPi will use it.

Note that if the panel chucks out a high voltage (which we have seen happen in practice) it will fry your Pi!

HTH, H

closed 2016-12-23 by hamishcunningham

ronisegoly, 2016-12-23:
I ran the board today with solar panel and battery. 
The RPI logs every 5 minutes to excel on Google drive, so I can see both outputs and which one is [active](https://docs.google.com/spreadsheets/d/1WhJ8PmZZpbeeKjkl9Qz67hIg4czKsv6qBtkDXewkgFo/edit?usp=sharing)
Indeed the MOPI choose the source with higher voltage, so even if the panel can still act as power source, if battery is fully charged it will switch to battery
I was hoping it will use solar whenever is is available