hamishcunningham/pi-tronics #104: mopicli. Invalid parameter, value outside range

McTucker, 2017-02-20:
Subject: mopicli:
-won seconds	Power on timer in seconds
-wsd seconds	Shutdown timer in seconds

Question:
What are the maximums for the settings -won and -wsd?

Problem:
now is 12:16:49
go_to_sleep at 11:56:00
wake_up at 11:56:02
So, I should give the parameters: mopicli -wsd 85151 -won 2
( One day has 86400 seconds ... ;-)
Output: "mopicli. Invalid parameter, value outside range"

stoduk, 2017-02-20:
Search for "value outside range" in the code, the value must be positive and below 0xFFFF.  This is 65535s or 18.2 hours.  [there *may* be further restrictions on the value on the MoPi side, this is just the client side/API restriction, though I doubt it]

closed 2017-02-20 by stoduk

McTucker, 2017-02-20:
Hi stoduk,

Thanks for your quick aswer.

I've found an other methode.
I build a time in my program, so I let the Pi sleep with -wsd 10 and weak him up after 8 hours sleep with 
-won 8*60*60

stoduk, 2017-02-20:
Glad you sorted it.  If you do end up needing to sleep for longer than 18.2 hours, then I think your only option is to do it in parts (eg. to sleep for 30 hours, sleep for 18.2, wake up and schedule sleep for 11.8 hours).  HTH