hamishcunningham/pi-tronics #107: Mopi hangs after about 30 days working.

McTucker, 2017-05-29:
At 23:00 mopi sets pi to sleep. at 07:00 is wakes up the pi again.
This works for about 30 days without any problems.
But after about 30 days, mopi doesn't wake-up the pi anymore. 
Then I have to disconnect mopi from power (12v car baterry) and then it will work for the next 30 days.

Is there a way to reset the mopi by software to avoid this problem?

Regards, Harrie.

hamishcunningham, 2017-05-29:
Hmm, that's a wierd one! Did you try reading status from the board and seeing if anything changes over the course of the runtime? Apart from that I guess an arrangement where pulling down the Pi supply also pulls down the MoPi supply for a few seconds would be fairly easy to come up with...?

McTucker, 2017-05-29:
I will check this (after about one month, when it's hanging again ;-)
Will let you know!
Thanks for your fast reply, and the fine mopies (yep, got two of this fine things)

closed 2017-05-29 by McTucker