hamishcunningham/pi-tronics #109: Communication error on read word

magusian, 2017-07-16:
After a Raspbian update/upgrade, I get the following error:

/usr/sbin/mopi: /usr/sbin/mopicli not working: installation problem? (mopicli. Communications protocol err on read word.)

2x MoPi v3.0
/usr/sbin/mopi version 4.1
2x Raspberry Pi 2 Model b v1.1
Raspbian GNU/Linux 8 (jessie), image dated 05/07/2017, Kernal version: 4.9 (virgin image)

I have tested all possible permutations based on 2x Pi 2's, 2x MoPi's, an upgraded and a fresh Rasbian install, but get the same error with all combinations

Can you suggest you to fix?

Regards,
Bradley

jjpesante, 2017-08-07:
also having this issue with fresh installs on a pi2, a pi3, and a zero W, 05/07/2017 kernel 4.9, however, topi does work when run with 2016 build of raspian as long as it is not updated. has anyone found a solution?

CosmaP, 2017-08-08:
I have the same problem with a Pi 3

SSDDscot, 2017-10-11:
i cant remember the link but i changed my config file to include the bcm2708. it goes back to a recent update i think.
sorry i cant be more help.

CosmaP, 2017-10-11:
Essentially it is not really supported on Pi3.

https://pi.gate.ac.uk/posts/2016/12/19/good-news-and-bad/

hamishcunningham, 2017-10-11:
The problem may not not the Pi 3 itself but the change in the way the
kernel does i2c in the more recent raspbian.

See #110 -- if other people could try the workaround cited there (which has
been reported once not to work)
and see if it fixes this issue that would be cool...

-- 

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK

*Phone: 07740 149 842 *
hamish@gate.ac.uk            https://twitter.com/@HCunningham
https://hamish.gate.ac.uk    *https://wegrow.social
<https://wegrow.social/>*
https://gate.ac.uk           https://pi.gate.ac.uk
*https://www.sheffield.ac.uk/dcs/people/academic/hcunningham
<https://www.sheffield.ac.uk/dcs/people/academic/hcunningham>*
https://www.sheffield.ac.uk/news/nr/aquaponics-fish-food-farm-1.581536
https://tinyurl.com/savetheworldwithfishpoo

WeGrow.social:
https://twitter.com/WeGrowSocial
https://www.facebook.com/WeGrowSocial
https://blog.wegrow.social
https://www.instagram.com/wegrowsocial
https://wegrow.social
https://www.youtube.com/channel/UCG3ZPbhbJ1yoa0fxMnaOcdg/videos

On BBC Radio Sheffield:

   - with John Grant: http://bbc.in/2ceFE57 (8 mins in)
   - https://hamish.gate.ac.uk/bbc-shef-aquaponics-may-2016.wav
   - https://www.facebook.com/bbcsheffield/videos/1214299238582354/

CosmaP, 2017-10-11:
> 335730731>,
> *Phone: 07740 149 842 <07740%20149842> *

I will give it a go.  But the mopi is actually doing what I need at the
moment.  That is the ability to headlessly                      start/stop
the pi cleanly without having to ssh and issue a shutdown command.

The rest of the functionality would be nice, but I can live without it.

When they release a Pi3 compatible version I will get one and move my
current one to a Pi1 that I am using as "house" Pi.

*________________________________*

*Cosma *

Mobile: 07956 323938
Skype: Cosma.Papouis

*P* *Please consider the environment*
*Save paper - do you really need to print this email?*

*Save electricity - do you really need to send this email? Save your own
energy - do you really need to read this email?*

CosmaP, 2017-10-13:
Do you mean this

 

https://pi.gate.ac.uk/pages/mopi.html#section-10.1.

 

________________________________ 

Cosma Papouis

Skype: Cosma.Papouis

e:  <mailto:cosma.papouis@uatechserv.com> cosma.papouis@uatechserv.com

 

hamish@gate.ac.uk <mailto:hamish@gate.ac.uk>  https://twitter.com/@HCunningham
https://hamish.gate.ac.uk *https://wegrow.social
<https://wegrow.social/>*
https://gate.ac.uk https://pi.gate.ac.uk
*https://www.sheffield.ac.uk/dcs/people/academic/hcunningham
<https://www.sheffield.ac.uk/dcs/people/academic/hcunningham>*
https://www.sheffield.ac.uk/news/nr/aquaponics-fish-food-farm-1.581536
https://tinyurl.com/savetheworldwithfishpoo

https://twitter.com/WeGrowSocial
https://www.facebook.com/WeGrowSocial
https://blog.wegrow.social
https://www.instagram.com/wegrowsocial
https://wegrow.social
https://www.youtube.com/channel/UCG3ZPbhbJ1yoa0fxMnaOcdg/videos

- https://hamish.gate.ac.uk/bbc-shef-aquaponics-may-2016.wav
- https://www.facebook.com/bbcsheffield/videos/1214299238582354/

Slewrate13, 2019-06-05:
Hi Everybody,
I'm trying to use extended feature of the Mopi1 I have buy some weeks ago. But I always get a "Communication protocol error on read word" as answer when trying to use "mopi" or "mopicli" command.  The hardware function seems to be all right.
Only "sudo mopi -d" is working. I've read no error message during installation. I've checked the I2C and I found BMC2835
I've tried with 2 different RPi zero and an old RPi B.
Sorry for my poor English. I've tried to follow #109 and #110 questions but I did not found the solution.
Best
Marc

hamishcunningham, 2019-06-12:
I think the workaround is described here: https://github.com/raspberrypi/firmware/issues/828
hth
h

closed 2019-06-12 by hamishcunningham