hamishcunningham/pi-tronics #110: cannot install simbamond on Stretch Lite

ArcticSnow, 2017-09-28:
Hi,

Just installed a fresh version of the official raspbian distribution of Debain Strectch Lite (https://www.raspberrypi.org/downloads/raspbian/), and when I run the command sudo apt-get install simbamond, it returns:

E: Unable to locate package simbamond

hamishcunningham, 2017-09-28:
Thanks, I've given Alex B a shout... will get back you soon...

hamishcunningham, 2017-09-29:
Serge S has kindly copied it in... does it work now?

ArcticSnow, 2017-10-01:
The install went through!  Thanks for the prompt help!

ArcticSnow, 2017-10-01:
but...

sudo mopi
[sudo] password for driftlidar: 
/usr/sbin/mopi: /usr/sbin/mopicli not working; installation problem? (mopicli. Remote I/O error)
Usage: mopi [-h(elp)] [-d(ebug)] [-v(ersion)] [-l(og entries)]

hamishcunningham, 2017-10-01:
did it work on the previous OS? is it powered up? (sorry to ask but...!)
is i2c enabled?
cheers
h

-- 

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK

*Phone: 07740 149 842 *
hamish@gate.ac.uk            https://twitter.com/@HCunningham
https://hamish.gate.ac.uk    *https://wegrow.social
<https://wegrow.social/>*
https://gate.ac.uk           https://pi.gate.ac.uk
*https://www.sheffield.ac.uk/dcs/people/academic/hcunningham
<https://www.sheffield.ac.uk/dcs/people/academic/hcunningham>*
https://www.sheffield.ac.uk/news/nr/aquaponics-fish-food-farm-1.581536
https://tinyurl.com/savetheworldwithfishpoo

WeGrow.social:
https://twitter.com/WeGrowSocial
https://www.facebook.com/WeGrowSocial
https://blog.wegrow.social
https://www.instagram.com/wegrowsocial
https://wegrow.social
https://www.youtube.com/channel/UCG3ZPbhbJ1yoa0fxMnaOcdg/videos

On BBC Radio Sheffield:

   - with John Grant: http://bbc.in/2ceFE57 (8 mins in)
   - https://hamish.gate.ac.uk/bbc-shef-aquaponics-may-2016.wav
   - https://www.facebook.com/bbcsheffield/videos/1214299238582354/

ArcticSnow, 2017-10-02:
Ok, so with a 12 v power supply the pi runs on the mopi just fine, but I obtain the same error.  I must run sudo mopi -d  to reach the configuration menu. Is this equivalent, or are these changes in debug mode temporary?

Thanks

hamishcunningham, 2017-10-03:
Hello again. I think this may be related to the I2C repeated start change in kernel versions that happened a little while ago. This thread https://github.com/raspberrypi/firmware/issues/828 describes a workaround that involves loading the older I2C module (or overlay) -- would you be able to try that and see if it works? Thanks! Hamish

ArcticSnow, 2017-10-05:
I followed these steps from one of the Ghost's comment on the thread you mentioned, without a chance:

    Download the old module here.
    Copy this file into /boot/overlays.
    In /boot/config.txt add the line dtoverlay=i2c1-bcm2708 at the end.

I am still getting the same error. The RTC (DS1307) runs also through i2c without an issue.
Thanks

ThomLinley, 2018-01-23:
I ran into this issue over the last few days. Using the Nov 2017 image of RASPBIAN STRETCH LITE on a B+ board the workaround from Ghost's comment worked for me. I am now able to use sudo mopi and communicate with the board. 

I was also able to use:
`sudo sh -c '/bin/echo Y > /sys/module/i2c_bcm2708/parameters/combined'`

to hopefully prevent conflicts down the line. I am also using a RTC and seem able to set that up using the newer overlay method.

Might be worth another shot ArcticSnow.

closed 2019-06-12 by hamishcunningham