hamishcunningham/pi-tronics #112: MoPi 3

CosmaP, 2018-04-01:
Hi, 

Where do we stand with a MoPi that works with a Pi 3?

The last thing I can find says it is almost ready, but that was about 3 years ago.

Thx

hamishcunningham, 2018-04-16:
It is basically ready for production... https://photos.app.goo.gl/DOuixIUBPhJykqfH2 :)

But the Pi 3+ came along with the PoE connector, which requires a little rejig to the PCB...

closed 2018-04-16 by hamishcunningham

CosmaP, 2018-04-16:
Excellent.

How much?

Can I have one please 😁 

---------------------------------------
Cosma Papouis
Mobile: 07956 323938
Skype: Cosma.Papouis

Please excuse abrupt emails. I am responding from my phone.

hamishcunningham, 2018-04-16:
> DOuixIUBPhJykqfH2 :)

They'll be on sale at pimoroni.com ASAP :)

-- 

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK

*Phone: 07740 149 842 *
hamish@gate.ac.uk            https://twitter.com/@HCunningham
https://hamish.gate.ac.uk    *https://wegrow.social
<https://wegrow.social/>*
https://gate.ac.uk           https://pi.gate.ac.uk
*https://www.sheffield.ac.uk/dcs/people/academic/hcunningham
<https://www.sheffield.ac.uk/dcs/people/academic/hcunningham>*
https://www.sheffield.ac.uk/news/nr/aquaponics-fish-food-farm-1.581536
https://tinyurl.com/savetheworldwithfishpoo
https://www.youtube.com/channel/UCG3ZPbhbJ1yoa0fxMnaOcdg/videos

On BBC Radio Sheffield:

   - with John Grant: http://bbc.in/2ceFE57 (8 mins in), or
   https://hamish.gate.ac.uk/bbc-shef-aquaponics-may-2016.wav
   - https://www.facebook.com/bbcsheffield/videos/1214299238582354/

CosmaP, 2018-04-26:
Great, Thanks