hamishcunningham/pi-tronics #115: Communication error on read word - stil not working on Rasbian

Slewrate13, 2019-06-13:
Hello Folks,
I'm still trying to use extended feature of the Mopi1 I have buy some weeks ago. But I always get a "Communication protocol error on read word" as answer when trying to use "mopi" or "mopicli" command. The hardware function seems to be all right.
Only "sudo mopi -d" is working. I've read no error message during installation. I've checked the I2C and I found BMC2835
I've tried with 2 different RPi zero and an old RPi B.  I've tried to follow #109 and #110 and #828 questions but I did not found the solution.

Indeed, I do not understand updated i2c0_bcm2708 firmware could be found.  i cannot understand what the combined parameter actually means. The discussion #828 is a bit confusing to me.  I cannot grasp what the solution is/was.

I added the following line into /boot/config.txt, then reboot

`dtoverlay=i2ci_bcm2708,dtbo
`

I tested the result with the following command 
```
pi@raspberrypi:~ $ mopicli -v1
mopicli. Communications protocol error on read word.
pi@raspberrypi:~ 

```

In order to take the exact version installed I entered fowolowing command 
```
pi@raspberrypi:~ $ cat /sys/module/i2c_bcm2708/srcversion 
0C1F4F8690E98B7CEAABAE5
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.46+ #1231 Mon Jun 3 17:28:02 BST 2019 armv6l GNU/Linux
```$ 
Thanks in advance for any help.
Marc

tennisparty, 2020-03-30:
Check out Adams pull request, if you update the mopiapi.py file with his suggested changes do you still get the error? [Adams pull request](https://github.com/hamishcunningham/pi-tronics/pull/118)

tennisparty, 2020-04-02:
I think you need to apply the workaround first:
https://github.com/raspberrypi/firmware/issues/828#issuecomment-311262108