hamishcunningham/pi-tronics #116: MoPi i2c Remote I/O Error

sinoia, 2019-08-05:
Hello

Please could you help with the issue I have - this is on a new MoPi 2 that I just got a few days ago. It is installed on a Pi 3  A+. After an initial success the board is not being recognised on i2c bus.

To start I installed all the software, attached batteries, wrote battery configuration and ran on battery for about an hour. All this went smoothly. Switched off using the power button, shutdown worked and all the LEDS went off. 

The next day ....

Pressed the button to start up but the pi would not start up. The two battery leds flashed green a few times and then went off. 

Powered up using micro usb and found the board not detected on i2c. 

Disconnected everything, retried, retried other i2c devices and they were all detected. But cannot get MoPi to be detected.

Various debug information below.

Thanks

**> pi@raspberrypi:~ $ sudo mopi**
> /usr/sbin/mopi: /usr/sbin/mopicli not working; installation problem? (mopicli. **_Remote I/O error_**)
> Usage: mopi [-h(elp)] [-d(ebug)] [-v(ersion)] [-l(og entries)]
> 
> 

**> pi@raspberrypi:~ $ sudo mopi -v**
> /usr/sbin/mopi is at version 4.1

**> pi@raspberrypi:~ $ sudo mopi -l**
> Aug  5 21:25:21 raspberrypi root: simbamon: ...boot delay done
> Aug  5 21:25:21 raspberrypi root: simbamon: writing local config /usr/sbin/mopicli -wc 2 9600 7400 5200 4800
> Aug  5 21:25:22 raspberrypi root: simbamon: error writing config: mopicli. Remote I/O error
> Aug  5 21:25:23 raspberrypi root: simbamon: MOPI_STATUS is mopicli. Remote I/O error
> Aug  5 21:25:23 raspberrypi root: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> Aug  5 21:26:34 raspberrypi root: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> Aug  5 21:27:35 raspberrypi root: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> Aug  5 21:28:36 raspberrypi root: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> Aug  5 21:29:37 raspberrypi root: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> Aug  5 21:30:38 raspberrypi root: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> 

**> pi@raspberrypi:~ $ cat /sys/firmware/devicetree/base/model;**
> Raspberry Pi 3 Model A Plus Rev 1.0

**> pi@raspberrypi:~ $ cat /etc/os-release**
> PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
> NAME="Raspbian GNU/Linux"
> VERSION_ID="10"
> VERSION="10 (buster)"
> VERSION_CODENAME=buster
> ID=raspbian
> ID_LIKE=debian
> HOME_URL="http://www.raspbian.org/"
> SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
> BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

**> pi@raspberrypi:~ $ service simbamond status**
> ● simbamond.service - LSB: simbamond: a simple battery monitor daemon
>    Loaded: loaded (/etc/init.d/simbamond; generated)
>    Active: active (running) since Mon 2019-08-05 21:14:35 BST; 3min 28s ago
>      Docs: man:systemd-sysv-generator(8)
>   Process: 503 ExecStart=/etc/init.d/simbamond start (code=exited, status=0/SUCCESS)
>     Tasks: 2 (limit: 860)
>    Memory: 5.9M
>    CGroup: /system.slice/simbamond.service
>            ├─535 /bin/bash /usr/sbin/simbamon
>            └─693 sleep 60
> 
> Aug 05 21:14:35 raspberrypi simbamond[503]: Started simbamon.
> Aug 05 21:14:35 raspberrypi systemd[1]: Started LSB: simbamond: a simple battery monitor daemon.
> Aug 05 21:14:35 raspberrypi root[537]: simbamon: first run after boot, sleeping for 2...
> Aug 05 21:14:37 raspberrypi root[541]: simbamon: ...boot delay done
> Aug 05 21:14:37 raspberrypi root[551]: simbamon: writing local config /usr/sbin/mopicli -wc 2 9600 7400 5200 4800
> Aug 05 21:14:39 raspberrypi root[553]: simbamon: error writing config: mopicli. Remote I/O error
> Aug 05 21:14:40 raspberrypi root[555]: simbamon: MOPI_STATUS is mopicli. Remote I/O error
> Aug 05 21:14:40 raspberrypi root[558]: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> Aug 05 21:16:52 raspberrypi root[645]: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> Aug 05 21:17:53 raspberrypi root[692]: simbamon: invalid MOPI_STATUS (mopicli. Remote I/O error): will pause/retry...
> 

**> pi@raspberrypi:~ $ i2cdetect -l**
> i2c-1   i2c             bcm2835 I2C adapter                     I2C adapter
> pi@raspberrypi:~ $ i2cdetect -y 1
>      0  1  2  3  4  5  6  7  8  9  a  b  c  d  e  f
> 00:          -- -- -- -- -- -- -- -- -- -- -- -- --
> 10: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 20: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 30: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 40: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 50: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 60: -- -- -- -- -- -- -- -- -- -- -- -- -- -- -- --
> 70: -- -- -- -- -- -- -- --
> 

> pi@raspberrypi:~ $ sudo lsmod | grep i2c
> i2c_bcm2835            16384  0
> i2c_bcm2708            16384  0
> i2c_dev                20480  0
>

hamishcunningham, 2019-08-06:
hi John

Thanks for the detailed report!

It sounds like either the board isn't powered, or not properly connected, or has suffered some catastrophic damage or failure. I haven't seen the latter happen before, so the likely suspects are the first two...

Hope that helps somehow!

Best, Hamish

sinoia, 2019-08-06:
Hi Hamish

Thanks for the quick reply and it did help - it made me think a little more!

I came home and connected 8xAA and everything started up perfectly - so just as you suggested the board was not powered. 

I discovered something interesting though - you probably already know this - 2x9V PP3 batteries with 8.5V charge is not enough to start the Pi. It is fine for running the Pi - although under load there would perhaps be problems. I know this because I started up the Pi with the 8xAA and then switched over to the two partly depleted PP3 cells and the Pi continued running. These were standard Alkaline 9V cells; I have a plan to try some 800mAh Lithium-Ion rechargable PP3 cells which I don't have yet and so cannot try!

Thanks for the help and for the great power board.

Rgs
John

closed 2019-08-06 by sinoia