hamishcunningham/pi-tronics #117: Starting up mopi2 with

afgoo, 2019-08-23:
Hi,

I've been having issues setting up my MoPi 2 with a raspberry pi 3.

I'm trying to use a 2 cell Li-ion battery to power it. I have finally gotten the pi to accept the battery settings and not restore the defaults on reboot.

I can power the mopi with a bench supply at around 9 V, and the light changes colour appropriately as I reduce the voltage. I can't power the mopi on with 7.4 V after it is shut down. It seems that the voltage settings are being stored on the pi, but not being stored on the MoPi itself, which prevents the board being powered up again on the lower voltage.

petaflot, 2020-08-13:
"I have finally gotten the pi to accept the battery settings and not restore the defaults on reboot."

that makes me believe the firmware should be open as well as the user-space utilities.

petaflot, 2020-08-15:
can you tell me how you achieved that? I didn't manage, after a reboot I always get the default values back.

afgaungoo, 2020-08-17:
Hi Petaflot,

I hope you can appreciate that it was about a year ago when I was last working with this, so my memory on it is quite vague. I don't recall how I finally managed to get the pi to store my settings, I might have located the mopi configuration file and edited it directly in a text editor. But these settings were only saved on the Pi itself, not the MoPi. So after I powered up with around 9V, and the pi booted up, the config file was read from the pi and I could drop the voltage down to 7.4 V before it shut down.

From what I understand, the MoPi has its own on board configuration, and also the pi stores a configuration when you change it using the software or editing the file. The MoPi updates its on board configuration to copy what's in the pi configuration file, but it can only do this when the pi is on. The MoPi seems to be unable to write to its non-volatile memory, so cannot preserve settings changes after a power cycle. This means that it always goes back to the default configuration after a power cycle, which, if I recall, is for a number of cells, requiring something like a minimum of 8.4 V to turn on. 

Unfortunately, this means that the MoPi could not achieve what I needed, and as I have had no response about this major issue, I have had to use something else. As my project is a commercial project, we ended up using expensive smart batteries.

Sorry I can't be more help.

petaflot, 2020-08-17:
that is actually massively useful (although it does not fix the problem)

 with what you describe, I can pretty much assume that simbamond reads the config from the file and quietly updates the MoPi internal config to pretend that was saved.

mopi does not even keep it's config when it powered and in a off state, having this major issue unresolved for over a year with not even an answer from Cunningham leads me to think this guy is just a fraud ; I will try to get my hands on the firmware to fix it myself.

have a nice day