hamishcunningham/pi-tronics #120: Manually entered voltage values are lost

tennisparty, 2020-03-30:
When I use the command 

> sudo mopicli -wc1 1 13000 12300 12050 11810

The board accepts voltages for source 1 (an AGM 12v battery), these can be seen by running 'mopicli -e'. When the board is reset this voltages values are lost (again seen by running 'mopicli -e'). The values are reset to those I have entered using the mopi configuration tool (sudo mopi - if anyone is struggling using this, you select values with the space bar!!). Values gained with the mopi configuration tool are not lost but I cannot enter manual values on this tool and so I have chosen some rather bizarre values which half match those I would like (an other - seven cell - lead acid). Do you know why the voltage values entered into terminal are being lost?

Many thanks,
Sam

hamishcunningham, 2020-04-01:
I think that one of the routes you're using writes local config into /etc and the other doesn't. The relevant code is: https://github.com/hamishcunningham/pi-tronics/blob/2ff38315db8281d4ad9406fc59649f3a42f140cc/simbamon/simbamond.default#L197

HTH, H

tennisparty, 2020-04-04:
OK great. I have cleared the settings in the MoPi app but the voltage settings are now defaulting to 11200 9800 9300 8500 ( I imagine the default mopi settings). If I delete the save local config line will my mopicli -e method values be retained. Otherwise I was thinking of deleting most of / edit the simbamond file to send out the values I need. Thanks for your help, Sam

tennisparty, 2020-04-05:
I decided to add an extra line in the simbamond.default file for AGM settings. I can't quite work out the calculations though and if I enter VHigh = 2.16 (13/6) and Operating floor = 2.03 (12.2/6) I keep getting an error message that the good battery level is lower than the low battery level

I decided to add some settings for AGM (Vhigh 2.16 and Operating floor 2.04) and edit the low battery multiplication setting to 1.02 - to avoid the error described above:
  BAT_FLOOR=$( echo "`bat_dod_floor ${CHEMISTRY}`*${NUM_CELLS}"          |bc )
  BAT_LOW=$(   echo "${BAT_FLOOR}*1.02"                                  |bc )
  BAT_FULL=$(  echo "`bat_nominal_volts ${CHEMISTRY}`*${NUM_CELLS}"      |bc )
  BAT_GOOD=$(  echo "scale=3;((${BAT_FULL}-${BAT_LOW})/2)+${BAT_LOW}"    |bc )
}

But even though my battery voltage is higher than the low battery level I am getting a rather persistent broadcast message - simbamon: status word is low! connect new battery or shut down

Is there something else I need to edit? Or a better way of calculating the values needed for an AGM battery?

tennisparty, 2020-04-05:
All good, I have edited the simbamond.default file - to enter the AGM values, I had to make some changes to the low battery multiplication settings to enter the values I needed. I kept getting an error about the good voltage being bigger than the low voltage because the calculations for low and good voltages are a bit tricky to work out.

tennisparty, 2020-04-06:
Hi Hamish,

I am sending poweron delays to the board through python. At what voltage level does the board not power up, is this at the low or the critical voltage level? I want to ensure the battery doesn't run when the voltage falls below 12.2V (c.25% full - to keep it in good working order). I am wondering whether I should set the **critical** or the **low** voltage level at 12,200,

Thanks again,
Sam

hamishcunningham, 2020-04-07:
IIRR the low voltage level triggers warning (via the LEDs) and the critical
level triggers shutdown
hth
h

-- 

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK

Cunningham's law of administrative entropy: the more data you enter, the
more you will be asked for.
Google's law of data value: if you can't get the machine to do it, don't
bother.

*Phone: 07740 149 842 / 0114 222 1891*
hamish@gate.ac.uk            https://twitter.com/@HCunningham
https://hamish.gate.ac.uk    *https://unphone.net <https://unphone.net/>*
https://gate.ac.uk           https://pi.gate.ac.uk
*https://www.sheffield.ac.uk/dcs/people/academic/hcunningham
<https://www.sheffield.ac.uk/dcs/people/academic/hcunningham>*

tennisparty, 2020-04-25:
Hi Hamish,

I am having an issue when the battery reaches c.1/3 of the discharge level. The simbamon logs at this point show the following:

> Apr 23 05:46:00 raspberrypi root: simbamon: BAT_WARNING (49169)
Apr 23 05:47:03 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:47:03 raspberrypi root: simbamon: BAT_WARNING (49169)
Apr 23 05:48:07 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:48:07 raspberrypi root: simbamon: BAT_WARNING (49169)
Apr 23 05:48:37 raspberrypi root: simbamon: status word is at or below warning level (49169)
Apr 23 05:49:11 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:49:11 raspberrypi root: simbamon: BAT_WARNING (49169)
Apr 23 05:50:15 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:50:15 raspberrypi root: simbamon: BAT_WARNING (52497)
Apr 23 05:51:19 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:51:19 raspberrypi root: simbamon: BAT_WARNING (52497)
Apr 23 05:52:10 raspberrypi root: simbamon: status word is 52497 (i=16000)
Apr 23 05:52:23 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:52:23 raspberrypi root: simbamon: BAT_WARNING (52497)
Apr 23 05:53:27 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:53:27 raspberrypi root: simbamon: BAT_WARNING (52497)
Apr 23 05:53:57 raspberrypi root: simbamon: status word is at or below warning level (52497)
Apr 23 05:54:31 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:54:31 raspberrypi root: simbamon: BAT_WARNING (52497)
Apr 23 05:55:35 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:55:35 raspberrypi root: simbamon: BAT_WARNING (52497)
Apr 23 05:56:39 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:56:39 raspberrypi root: simbamon: BAT_WARNING (52497)
Apr 23 05:57:39 raspberrypi simbamond[301]: wall: cannot get tty name: Inappropriate ioctl for device
Apr 23 05:57:39 raspberrypi root: simbamon: shutting down (POWER_OFF; 57745)
Apr 23 05:57:39 raspberrypi systemd[1]: Stopping LSB: simbamond: a simple battery monitor daemon...
Apr 23 05:57:44 raspberrypi systemd[1]: Starting LSB: simbamond: a simple battery monitor daemon...
Apr 23 05:57:44 raspberrypi root: simbamon: starting...
Apr 23 05:57:44 raspberrypi root: /usr/sbin/simbamon: version 4.1 running at Apr-23-2020-05:57:44
Apr 23 05:57:44 raspberrypi root: simbamon: monitor frequency is 2 seconds
Apr 23 05:57:46 raspberrypi root: simbamon: first run after boot, sleeping for 2...
Apr 23 05:57:46 raspberrypi simbamond[294]: Started simbamon.
Apr 23 05:57:46 raspberrypi systemd[1]: Started LSB: simbamond: a simple battery monitor daemon.<

The board was meant to turn off at 05:57 on Apr 23 then reboot at dusk (by a power on delay). It had worked for the previous 6 days but not when the BAT_WARNING showed error message (52497). I have been logging the voltages (only once every c.12 hours), at 05:50 on Apr 23 the voltage was 12337. I had set the voltage levels as follows - 
Max 14000 mV
Good 13000 mV
Low 12100 mV
Critical 11000mV
I would therefore expect the board to still be thinking the battery is low but not critical.

Is there a way to interpret the BAT_WARNING numbers. I have tried looking in the simbamon file but dont understand where they originate. I wonder whether the BAT_WARNING (52947) means the battery is considered to be at a critical voltage level. If so do you think the battery levels could be tailing off at this point and quickly transitioning into critical voltage. I thought this might be a reason why the power on delay is not being set on the mopi? Strange though, I powered the board on later that evening (with the power button) - I recorded some voltages on three individual boots - 12363, 12363 and 12317. These voltages are all still above the critical voltage level so it doesn't appear to be at a critical level. I didn't get a battery warning message on any of these mini boots but did get the log, if it helps diagnose anything:
>Apr 23 22:29:36 raspberrypi root: simbamon: MOPI_STATUS is 61449<

Any ideas would be much appreciated,
Many thanks,
Sam