hamishcunningham/pi-tronics #122: Mopicli, how to determine active source

azetat, 2020-08-04:
Hi, I'd like to use a cronjob/shell script to detect the MoPi-2 running on battery (supply 2), if mains failed (supply 1).
Is there any way (version 4.1) to get the active source using mopicli?
Thanks, Volkmar

hamishcunningham, 2020-08-04:
You should be able to figure it out from a combination of -v (current source volts) and -v1 -v2 (the individual supply volts).
HTH
H

azetat, 2020-08-04:
Thanks, got you. So no direct parameter...

hamishcunningham, 2020-08-04:
not that I remember; the options are listed if you do -h
h

-- 

*Hamish Cunningham*
Professor of Computer Science, University of Sheffield, UK

Cunningham's law of administrative entropy: the more data you enter, the
more you will be asked for.
Google's law of data value: if you can't get the machine to do it, don't
bother.

*Phone: 07740 149 842 / 0114 222 1891*
hamish@gate.ac.uk            https://twitter.com/@HCunningham
https://hamish.gate.ac.uk    *https://unphone.net <https://unphone.net/>*
https://gate.ac.uk           https://pi.gate.ac.uk
*https://www.sheffield.ac.uk/dcs/people/academic/hcunningham
<https://www.sheffield.ac.uk/dcs/people/academic/hcunningham>*

closed 2020-08-04 by azetat