hamishcunningham/pi-tronics #123: MoPi won't boot.

petaflot, 2020-08-10:
Got 3 MoPi2 from Pimoroni, two have been tested and exhibit the same behaviour. Have not taken the third out of the packaging yet.

When power switch is pressed, green LED corresponding to the connected battery (or batteries when both are connected) light for some two seconds, then off again. No power is ever output to the the Pi, whether a Pi is connected or not (measured with multimeter). Battery voltage has also been tested, as well as power supply when connected.

MoPi2 is powered with one or two 9V batteries, connected with the provided connectors. Batteries are brand  new. 

Tested with and without external power supply (recycled laptop power supply 20V / 4A ; power supply voltage was known to work and was also tested with voltmeter.)

hamishcunningham, 2020-08-10:
First thing that comes to mind is that the Pi is not situated properly, or that there has been a short on the output pins. Another thing to try would be to check that MoPi is configured to expect a 9V supply, and not shut down due to thinking that it has met the minimum discharge voltage of the default batteries (8x NiMH). If your 20V supply is working that seems unlikely though. What happens when you "sudo mopi"?

petaflot, 2020-08-10:
No Pi attached means there _cannot_ be a short (unless it's on the MoPi itself). When the Pi is plugged on the MoPi (or the opposite, however you prefer to see it) the behaviour is the exact same. The Pi is brand new and works as expected (tried with Pi3B, Pi3B+, Pi4)

I cannot `sudo mopi` since the Pi doesn't boot at all. Also, I am running either gentoo or arch on my Pi because honestly debian derivates barely deserve to be called "linux OSes".

Also, here are some excerpts of my e-mails with the Pimoroni team:

> with or without any load, D7 NEVER lights up.

> I actually measured the voltage between pin 6 and pin 4, WITHOUT LOAD, WITHOUT POWER SUPPLY (on a single brand new 9V battery with the provided connectors) and the voltage stays at zero volts when the power switch is pressed (and D8 or D9 light up for 2 seconds). D7 still does not light up.

> From the datasheet[1] of the battery, "not very long" should be at least 7.5 hours with two batteries ; this is WAY ENOUGH TIME for the Pi to shut down. (according to Pimoroni, default profile is 8 AA batteries ; they suggested I should change the profile but how can I do this if the Pi doesn't boot?)

> is it acceptable to power to Pi through the USB port (standard) and, in this case, can we read/write data on the MoPi through the I²2C bus?

petaflot, 2020-08-10:
Oh, and I sent the Pimoroni team a picture of how I mounted the Pi on the MoPi and they had nothing to say about it.

hamishcunningham, 2020-08-10:
I recommend trying Raspbian just to be sure that there's not some problem in the OS.

petaflot, 2020-08-10:
1) I WAS NOT ABLE TO EVEN BOOT THE PI. As stated earlier, there is NO VOLTAGE AT ALL on the power pins, WITH OR WITHOUT a Pi attached. Should I try with a simple resistive load? If yes, what value do you recommend?

2) I'm pretty sure I actually tried with an SD flashed with a Raspbian a friend lent me for the purpose. No success either. The SD card was known to work.

hamishcunningham, 2020-08-10:
Right, but if you run the Pi from a USB power supply you should be able to talk to the MoPi board, configure it and so on?

petaflot, 2020-08-11:
Thank you, I will try this. Since some devices cannot stand being reverse-powered, I did not risk bricking the device.

petaflot, 2020-08-11:
Anyway, I tried.

Instead of getting through the hassle of setting up the network and installing the mopi executable, I simply ran

# watch -n 1 "i2cdetect -y 0 && i2cdetect -y 0"

then pressed the button a few times to see the MoPi is not detected. I also attached a single battery so I had some LED feedback, no change.

petaflot, 2020-08-14:
seems to totally relate to issue #125.

is X1 even connected to anything?