hamishcunningham/pi-tronics #124: Mopi 2 Py Zero W

kibroutski, 2020-08-11:
Hi there, 

Just received my mopi 2 from Digital-Key and I have it working fine on a Raspberry 3 B+. Now when I  test it on my Pi Zero W (Buster lite / I2C enabled / same power supply) I have the following error.

> pi@raspberrypi:~ $ sudo mopi
> /usr/sbin/mopi: /usr/sbin/mopicli not working; installation problem? (mopicli. Remote I/O error)
> Usage: mopi [-h(elp)] [-d(ebug)] [-v(ersion)] [-l(og entries)]

I did look in the previous errors and tried the workaround suggested in Issue #110 with no success. The board does power the Pi without any issue, I simply cannot run mopi. 

Any suggestions ?

closed 2020-08-11 by kibroutski

hamishcunningham, 2020-08-11:
I guess you got it running...?

kibroutski, 2020-08-11:
I'll try it tonight and update if it works fine on the Zero
It was late last night and I forgot the obvious check before posting the
issue