hamishcunningham/pi-tronics #125: Mopi doesn't boot with less than 12V

kibroutski, 2020-08-13:
Hi, this time I've done my due diligence and I'm almost certain that I haven't missed anything obvious. 

My initial goal was to power mi Pi using a 2 cells 7.4 V Li-Ion. To configure the Mopi, I use a variable power suply connected to X2 # 1 and managed to power the Pi+Mopi at 12 V. I configure using sudo mopi and reboot for good measure, after reboot my values are still present in the config file. 

Here comes the strange part. I disconnect X2 # 1 and connect the (charged) battery on X2 # 2 and the Mopi blinks green for a few second before shutting down. I tried to swap for a 9 V battery, reconfigure and same problem, the Mopi doesn't boot up. What's even stranger is that when I'm using both # 1 @ 12 V and # 2 @ 7.4 (or 9 v)  both D8 and D9 are solid green and if I disconnect # 1 then the Mopi continue to run and I can even reboot the Pi without any problems.  
In summary, I can't boot the Mopi with less than 12 V (might be lower but I can't test between 9 and 12) but once booted using a 12 V suply then the Mopi does run on lower voltage sources

The problem sounds similar to #117 and potentially #123 

I'm using Raspbian and I can reproduce the problem on a Pi Zero and a 3 B

petaflot, 2020-08-13:
Is your 12V power supply on X1?

I have 20V there and not much happens. I have no variable power supply to perform other tests.

antvicapp, 2020-08-13:
No on X2 
No success using X1

kibroutski, 2020-08-13:
No on X2
No success using X1
Your 20 V is on X1 ? If have you tried on X2 ?

If it does work on X2 then you might have the same problem

petaflot, 2020-08-14:
yup!

I scavenged an old linksys 12VDC power supply and now MoPi boots

kibroutski, 2020-08-14:
OK and just to confirm, once the Mopi is up did you config the power supply using sudo mopi ? If yes, are you able to boot up the mopi just from the 9 v battery ? 

On my side still no success booting the Mopi with anything else than 12 V.

petaflot, 2020-08-14:
duh.

I disconnected the 12V power supply on X2 by mistake, D7 started blink red, D8 and D9 started blinking too.

I reconnected the 12V power supply before Pi was shut down, but D7 was still blinking red and brutally shut down the Pi.

I was in the process of downloading mopi utilities, have to reboot and hope I don't have nasty filesystem corruptions.

kibroutski, 2020-08-14:
Ouch ok hope your SD is fine. 
Let me know if you have any success booting with the 9 V battery

petaflot, 2020-08-14:
having a hard time configuring my 12V power supply :+1: 

HC has very bad humour indeed

kibroutski, 2020-08-14:
Strange, you get this error using sudo mopi ? Also your 12 V is on X2 # 1 or # 2 ? Once thing that took me a while to figure out is that the config wizard uses old school Linux selection and to move the * you need to use the space bar

petaflot, 2020-08-14:
yeah, the mopi UI is so bad it is unusable. also had to fix the code at some places for mopicli to work : it relies on RPi.GPIO version  to decide whether I²C bus is 1 or 2. can you believe that?

got my way around using mopicli, guessing the values for a fixed 12V power supply:

# ./mopicli sw2 12000 12000 11800 11500

yep, having to use the space bar to move the * is quite confusing.

diplomacy prevents me from saying what I really think.

petaflot, 2020-08-14:
.. and still no clue about whether X1 is being used or not, what is the voltage there, what it's doing and where it's at.

FULL SCHEMATICS would be appreciated (as well as firmware source code and instructions to flash it)

kibroutski, 2020-08-14:
Yeah I was also confused to see that it was still using the old RPi.GPIO 
I've got a similar config for my 12 V except I did put some head room for max voltage 12.2 V 

As for X1 I contacted one of the board designer and here is his response 

> generally speaking MoPi should be connected as follows.
Connector X1. External suitable charger. In your case charger, able to charge 2 serially connected cells Li-Ion batteries.
Connector X2. Site #2 - 2 cells Li-Ion batteries, serially connected. Site #1 - another power source with voltage magnitude higher than the batteries voltage. This source will power the Pi during the batteries charge. The voltage magnitude should be at least 2V higher than the batteries max voltage, e.g. 12V DC wall adapter, or some solar panel or another battery pack.

petaflot, 2020-08-14:
oh, that explains a few things. that info is totally missing from the public description that leads to the belief X2 sites #1 and #2 are equivalent.

petaflot, 2020-08-14:
as soon as I have finished my system upgrade, I will sync my filesystem and try unplugging various things to see how it behaves.

kibroutski, 2020-08-14:
Yeah those info are pretty important info and yet missing, 
In any case, we're a bit digressing from the original issue here. I still can't figure out how to boot the Mopi with less than 12 V although the specs says that it should work from ~6 to 24 V and it's becoming really frustrating. 

If you can confirm that you have a similar behavior (i.e. a 9 V doesn't do the job although you configured it in the file system) that would be very helpful

petaflot, 2020-08-14:
well for starters mopi UI say max voltage is 25V.

(still doing my upgrade, running a gentoo from sakaki)

petaflot, 2020-08-15:
I can totally confirm that the mopi DOES NOT keep its config when it shutdowns ; config always goes back to the default so its no wonder it doesn't boot. My battery has a little over the low voltage setting, but when the Pi boots this will obviously drop somewhere between _low_ and _critical_.

MoPi did not even power on by itself after setting the delay with -won. shutdown worked more or less as expected (would it not be for the 30 seconds during which it remains on with blinking red D7 before actually shutting down - if you ask my that's way too long and _not_ the way to do it anyway).

Can you check your config is kept after a restart? I will now try with a second MoPi to see if it exhibits the same behaviour.

petaflot, 2020-08-15:
yeah, same thing. no hope there until "the man" comes back.

kibroutski, 2020-08-15:
I can confirm that by using sudo mopi to check, my config file is still the same after a reboot.
What command you use to check your config?

petaflot, 2020-08-15:
I just use mopicli, which uses mopiapi.py

the command I use is

# mopicli -e

so it doesn't check a config file, it pulls the values straight from the mopi hardware through I²C

petaflot, 2020-08-15:
your last comment might pretty well relate to #128 

as far as I can see, mopi UI writes the config to a file and uses this (instead of what is actually stored into the MoPi device) to show something to the user

kibroutski, 2020-08-16:
Thanks for the confirmation. OK so it seems all of our problems are converging towards the config file not being properly written/read and that it would lead to the Mopi always expecting slightly over 9V to boot (default settings).

petaflot, 2020-08-16:
I suppose you're selecting "read local config" after you have executed "sudo mopi". is this correct?

that "local config" is just as confusing as having to move the * with the space bar before hitting enter. I'm now pretty sure it actually stores the settings you selected with "sudo mopi", NOT what is actually being used by the device.

please confirm that your config is correct with

$ sudo mopicli -e

if this doesn't work because the path is not found, you may want to try

$ which mopi

this will tell you the full path of the mopi UI and you can pretty much suppose mopicli will be in the same directory. or, use a tool like

$ locate mopicli

let me know if you have an issues.

kibroutski, 2020-08-16:
Not a home at the moment so cannot test but  I did use both the "read local config" and the sudo mopicli -e and both were fine (i.e. right value) but yet as it seems clear now these are not the value used by the board

petaflot, 2020-08-17:
from #117 it became pretty clear the MoPi is unable to store its settings when shut down (even if powered).

this means it will _always_ use the default values when in an off-state ; this will prevent you from starting it with a voltage lower than 9.3V (not tested, I suppose it will expect a voltage equal to or above the "low voltage setting") or maybe 9.8V. this will also prevent you from starting the MoPi with the power-on delay.

my workaround was to order 8-cell NiMH battery packs because this is the only way to make sure the MoPi won't damage the batteries and will still boot ; I would have preferred another type of battery but hey - what the heck.

Now I just hope the MoPi will be able to charge those batteries with a power source connected to X1.

kibroutski, 2020-08-17:
On my project I don't have the luxury of space so I guess I'll have to return this board and get an Adafruit Powerboost + custom PCB to handle safe switch on/off 

It's too bad, with a few adjustments, this board could be great

muttsoft, 2020-08-25:
I'm having the problem where the MoPi 2 doesn't boot when connected to 7.6 v (LiPo in storage v.) or 8.4v.  The D8 or D9 just blinks a few times and it shuts down.  Both supplies are configured for a 2S LiPo.  Once the RPi is booted using 12v, I can hot swap to a 2S LiPo battery and run off of that just fine.  It seems like the MoPi isn't loading it's stored config until after it's booted.

petaflot, 2020-08-25:
muttsoft : so it seems, see #129 ; if that's the case you will have no luck whatsoever to boot unless you have almost 10V. 

car you try to disable the simbamond service / daemon and see if the conf is still restored on the MoPi?

petaflot, 2020-10-05:
yo!

"good" news here. I received my 8-cell NiMh batteries. See my last post on #129 for details.

I'm not wasting any more of my time with this crap.